theyosh/TerrariumPI #370: Cannot connect since update 3.9.8

jeff26300, 2020-04-06:
Hi theyosh,

I have just updated my TerrariumPi with version 3.9.8, by following your procedure, and I cannot connect to the TerrariumPi webpage anymore. I can still communicate with the raspberry via the terminal and after reboot I get the screen attached with the text "version 3.9.7". I reinstalled a second time and the same screen is shown, with the impossibility to connect to the webpage.
Am I missing something?

[image]

Thanks,

Jeff

theyosh, 2020-04-07:
Hi @jeff26300 
normally you have to do:
- stop terrariumpi: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi
- do a `git pull`
- rerun the installer: `sudo ./install`
- reboot just to be sure.

If that does not work, try debugging: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi and make a screenshot of what you see.

jeff26300, 2020-04-07:
Indeed it now works better :)

Thank you again for your great work!!

I can close the issue

closed 2020-04-07 by jeff26300