theyosh/TerrariumPI #371: Questions

martynwheeler, 2020-04-12:
Hi, this looks amazing. Would it work with an original raspberry pi? I assume that it's okay to use energenie sockets with a pi-mote?

theyosh, 2020-04-13:
Hmm, I would suggest not to use a Pi1. As it is pretty slow, I am afraid that installation will take about 2 -3 hours. And not sure if all the software will work due to old versions.

So I would suggest to use a new Pi. Either 3 or 4 is good. 

I have no experience with the pi-mote. I just added support based on other people input.

martynwheeler, 2020-04-13:
Yes, the Pi Model 1 is too slow. Just had one lying around and wondered if I could make use of it. Have a Pi3 setup and running with it now. Looks really good. Waiting for my Pi-mote to arrive but it looks like it should work having read through your code.

Really nice piece of work.

One final question, could I run another device on the I2c bus without it interfering with terrariumpi? I was thinking of putting a spare LCD display on the I2c bus to display the tank temperature for a quick visual check without having to look at the Web interface. This would take the data from the same temperature sensor in a cron script running once every couple of mins.

theyosh, 2020-04-15:
> One final question, could I run another device on the I2c bus without it interfering with terrariumpi? I was thinking of putting a spare LCD display on the I2c bus to display the tank temperature [...]

That should not be a problem. As long my software does not know about it, it will just leave it alone. So that should just work. Maybe handy to have: https://github.com/theyosh/TerrariumPI/wiki/API

martynwheeler, 2020-04-15:
Thanks,

I wish I'd seen that earlier.  I have looked at the db and cfg files and worked out where to get the data from.  Might look at the api though.

theyosh, 2020-04-15:
I would still go for the API with https://2.python-requests.org/en/master/

Then you need about 2-3 lines of code to get the right value from the API :)

martynwheeler, 2020-04-28:
Yes, I have done this now and it works brilliantly.  Thanks again.

closed 2020-04-28 by martynwheeler