theyosh/TerrariumPI #372: Unable to access TerrariumPI

justin1977justin, 2020-04-15:
I will start by saying that I am pretty new to RaspberryPI and Raspbian. That said, I have 4 other Raspberry PI setup and controlling things throughout my house. I am trying to install TerrariumPI but cannot access it after installation. I type ip_address:8090 and I get "This site can't be reached" "ip_address refused to connect". I can ping the Pi from my computer. Any suggestions on what may be wrong or how I can troubleshoot this? I've tried it multiple times on different PI, withe different versions of Raspbian, and different sd cards.

Current:
Raspberry PI Zero W
Raspbian Stretch Lite

martynwheeler, 2020-04-15:
Can you SSH into the pi? If so you can check if the service is running.

justin1977justin, 2020-04-15:
@martynwheeler I am able to ssh. I am not sure how to check if the service is running. I tried "systemctl is-active TerrariumPI.service" and it shows as inactive. Was that correct? If not, how would I check?

nastea1, 2020-04-15:
Check the processes for it. It might be running off the start.sh via rc.local. If that's the case, you can type screen -r and see what the log is saying. Just remember to hit CTRL-A and then D to detach.

justin1977justin, 2020-04-15:
@nastea1 I typed screen -r and it says "There is no screen to be resumed". I do not see any signs of it running. However, i may be doing it wrong.

theyosh, 2020-04-15:
Hmm, sounds like a failed installation. Also you are using a Pi Zero, which is a bit special.

So, I would do the following (I hope this will help):

`sudo -i` 
(make sure you are user root)

`cd /root` 
(should take you to the /root home folder)

`git clone --recursive --depth 1 https://github.com/theyosh/TerrariumPI.git`

` cd TerrariumPI`

`sudo ./install.sh 3`

This should re-install it as user root, which is needed for the Pi Zero. And for now, I know that Python2 has problems at the moment due to a just upgraded external library.

theyosh, 2020-04-15:
Also, if Debian buster is available for the Pi zero, use that.

justin1977justin, 2020-04-15:
Ok. Thanks. I will try that. I tried it on a 3B+ as well and had the same result. I was originally using Buster and was getting owserver errors and didn't know if it was related so I switched to Stretch which did not give the errors.

theyosh, 2020-04-15:
hmm, if you have again problems with owserver, could you send me the screenshot? I have purged it and then reinstalled it. No issues... and this is on Buster

justin1977justin, 2020-04-15:
The install finally finished and when i run screen -r it says   
"info - terrariumWrapper      - starting TerrariumPI server at"
when I try to connect from browser i still get refused to connect

justin1977justin, 2020-04-15:
I rebooted and can see the following TerreriumPI running that wasn't running previously

   481 ?           Ss      0:00 SCREEN -dmS TerrariumPI ./start.sh 2 run
 2554 pts/0    R+     0:09 python /home/pi/TerrariumPI/terrariumPI.py

theyosh, 2020-04-17:
Ok @justin1977justin 
I think it is wise to start over. As you have now a mixed setup with Python3 and 2. That will not work.

So, format a new SD card with Debian buster (raspian) and follow the steps at:

https://github.com/theyosh/TerrariumPI/issues/372#issuecomment-614126633

justin1977justin, 2020-04-17:
I did a fresh install again with Pi Zero with Buster Lite and got the owserver errors again. I then did a clean install on a 3b+ using your method (maybe not root, can't remember now) and finally got it to work. I've done a fresh install probably 10 times so i was about ready to quit. Thanks for your help.

theyosh, 2020-04-17:
So, now you have a Rpi 3b+ with buster? Did you use the Python3 install setup? added the '3' to the installer command?

justin1977justin, 2020-04-17:
I was able to get it working last after trying your suggestions several times. What finally worked was Full version of Buster on 3b+ using the install command you suggested. I also said no to removing unneeded items. Thanks for your help. I was about to give up.

justin1977justin, 2020-04-18:
Yes. The 3b+ is what worked. I had tried it previously as well and it did
not work. I also tried the Pi Zero W several times with your suggestion and
nothing worked. This working solution is on Pi 3b+ with Full Buster on
python 3 install. I don't think I did it as root but i may have. All my
attempts are running together at this point.

I am still getting errors where on boot. Attached is a photo of what
im talking about. Doesn't seem to be affecting anything that i can see.

On a side note, is there a way to reduce the temp sensor refresh rate for
heater control? At least for testing it would be nice to have it more
responsive.

Thanks for your help

theyosh, 2020-04-18:
Hmm, I think that the foto is lost in the email? I do not have a photo with this message. I am still interested what you get.

justin1977justin, 2020-04-19:
Sorry. I forgot to attach and then i got kicked back for being too large. 

[https://photos.app.goo.gl/Z8xd5q9yLvqVrT577](url)

justin1977justin, 2020-04-19:
Apparently link isn't working either. At least not on my end.
[image: 20200416_081639.jpg]

theyosh, 2020-04-28:
Are there still issues? Can we close this issue?

justin1977justin, 2020-04-28:
Not related to this issue so yes it can be closed. Thanks for your help.

closed 2020-05-01 by theyosh