theyosh/TerrariumPI #373: None% from Temperature and Humidity

nastea1, 2020-04-15:
After I leave my server up for some time, usually 3-7 days the DHT22's start throwing errors of none%. Usually getting in and restarting the daemon will fix it. Sometimes it requires a system reboot. Just the log shows received none% etc.

I have them wired with resistors and the normally read fine...just sometimes after a few days they stop. 

Let me know if there's any ideas.

theyosh, 2020-04-15:
Hi,

replacing with better hardware. My experience is bad with the DHT22 sensors. https://github.com/theyosh/TerrariumPI/wiki/Hardware#humidity And the BME280 is supported and has humidity support.

Or, you could also try https://github.com/theyosh/TerrariumPI/wiki/Hardware#gpio-power-saving .Then the sensor will be powered down, and that could maybe help in your case.

Else I do not have a solution: https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+dht22

nastea1, 2020-04-15:
Thanks. I'll try the power saving option and if not just go with higher end sensors.

closed 2020-04-15 by nastea1

jaysalazar, 2020-05-09:
@nastea1 did the power saving option work for you? I have the same problem DHT22/AM2302 are highly unstable from my expeirence.

nastea1, 2020-06-03:
> @nastea1 did the power saving option work for you? I have the same problem DHT22/AM2302 are highly unstable from my expeirence.

@jaysalazar Sorry for late response been busy with other projects, However no it didn't work.

theyosh, 2020-06-03:
Ok, final conclusion: Do not use DHTXX sensors. Use any other humidity sensors https://github.com/theyosh/TerrariumPI/wiki/Hardware#humidity