theyosh/TerrariumPI #374: password change not being saved across a reboot

martynwheeler, 2020-04-16:
Hi,

When I change the admin password it does not appear to be saved once the system has been rebooted.  I am putting in a new password and the existing password into the two fields.  Where is the hashed password saved in the system?

Thanks

Martyn

theyosh, 2020-04-17:
ah, that is wrong :P

You need to enter your CURRENT password and a NEW password in the fields. You can only change the existing password when you know it.

Password is stored in PLAIN TEXT in settings.cfg file. So it is not that secure. If you want more security, add a NGINX caching proxy in front.

martynwheeler, 2020-04-18:
That fixed it.  The input boxes are labelled the wrong way around in the gui! It asks for new password first and then the current password.  I could fix for you if you don't have time?

theyosh, 2020-04-28:
I do not understand? You think the fields are in an unlogic order? I think it does not matter that much. Nobody remarked it the last 4 years....

martynwheeler, 2020-04-28:
[image]

I think the boxes are wrongly labelled.  I think the first one is the current password and the second is the new password.  When I entered the values like that it all seemed to work.

theyosh, 2020-05-20:
Nope, fields are correct. Just tested. Closing issue.

closed 2020-05-20 by theyosh