theyosh/TerrariumPI #375: Use arduino as sensor source

martynwheeler, 2020-04-18:
How would you feel about adding the arduino as a source for sensors.  It is perhaps more flexible than the pi as a data collector and would mean that the GPIO pins were not tied up (so I could put my Pi-Mote hat directly onto the Pi).  It could add more versatility to the software.  I would be more than happy to help.

When adding new sensors, you would have an option for arduino, the sensor address could be a pin number on the arduino and then the type could be one of the already defined types.  It would be fairly simple to communicate using pyfirmata2 which can connect to an arduino using the USB on the pi.

Or is this already possible using remote sensors?  I can't see how to add multiple data from a serial remote source.

theyosh, 2020-04-18:
There is an option to load external data from external sources. That can be done in multiple ways. But I am not an experienced Adruino programmer, so it could be that some solutions are a bit hard to make.

First you can use https://github.com/theyosh/TerrariumPI/wiki/Remote-data . Your Arduino should run a webserver and have network access. Then you could make some small webserver on the Arduino that provide the needed measured values in JSON.

Or, somehow write the output from the Arduino to the Pi. And then write it in the static folder, so you can then use the TerrariumP webserver to load the data as remote data as describe above here.

Or, you can create a script in the Pi, that is able to get the measurements from the Adruino and use the script sensor type. And as you mention pyfirmata2, I think this is the easiest way.

So, if you can make a script in the Pi, that can readout the Adruino and just shows the readout value, you can then use this as a script sensor. Because you can do a lot of things this way, it is way to complex for me to implement all the options and features. Also, I do not have adruinos so testing would be hard.

theyosh, 2020-04-28:
Is the above explanation clear? As I would like to close this issue, as I cannot support more then what I have explained above. And I cannot support all kinds of external code.

martynwheeler, 2020-04-28:
Yes, this is great.  I have hooked up an arduino as a remote sensor.  As you say no need for more code.  Thanks again.

closed 2020-04-28 by martynwheeler