theyosh/TerrariumPI #376: Environment - Lights not remaining on when running on timer/weather

martynwheeler, 2020-04-18:
Hi,
Sorry for all the questions.  The lights are switching on and then off again repeatedly when they are supposed to be on.

My settings for "lights" are:
Mode: timer
Day
Start: 08:00 AM
Start: 08:00 PM
Timer on *: 720
Timer off *: 0
Power Switches: Lights
Night
Start: 08:00 PM
Start: 08:00 AM
Timer on *: 0
Timer off *: 720
Power Switches: Lights

Power switch for Lights is a Pi-Mote.
It just keeps toggling on and then immediately off again every 15 seconds.  Any ideas 

BTW, the heater is also on the Pi-mote and is working fine in sensor mode.

theyosh, 2020-04-18:
Hi @martynwheeler 

No problem! Solution is simple. Do not select the light switch at night time. Because during the day you want the light to be on at a period. That means outside that period the lights will be off. So no need to specify that at the night part.

The night part is there for IR lights for heating in the night or creating some moonlight effect.

So, just remove the lights power switch at the Night part, and it should work

martynwheeler, 2020-04-18:
Thank you! That worked.

theyosh, 2020-04-21:
Hi @martynwheeler 

No, the service is not available with a normal install. You have to do that manual. As the installer will make a screen session in the /etc/rc.local there is no need for the service. But if you want to use the service, you have first to manually comment out the screen line in /etc/rc.local and follow the steps here: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-systemd

theyosh, 2020-04-28:
Hi @martynwheeler can we close this issue? As your last question is actually a different issue. If that is still an issue I would like to ask to open a new issue.

closed 2020-04-28 by martynwheeler