theyosh/TerrariumPI #377: TerrariumPI on Pi 4

cyber-shifter, 2020-04-19:
So I saw this issue: https://github.com/theyosh/TerrariumPI/issues/372

I recently bought a new Pi with the intention of making an environment control system for my iguana enclosure. I did not realize a few things, as I am very new to raspberry pi. The 4 seemingly cannot run Stretch at first glance, but this software is not tested on Buster. I didn't realize this before I purchased the unit. SO, since I have this hardware, is there a way I can test it to help?

I had a fresh install of Buster lite on my device, but I could not get TerrariumPi to work after install. The service was not in the list of installed services as far as I could see, so that explains why it didn't start. I haven't found methodology for running this on Buster or on the 4 so I guess that's why I'm reaching out. Is it worth doing with the new hardware? Or should I just get a 3b+?

theyosh, 2020-04-19:
Hi @cyber-shifter 

Maybe the documentation is a bit outdated, but the software will run on a Pi 4 with Buster. I am currently running on Pi4 with buster in production. So that should not be an issue.

The only thing that is rather important, is that you install it with Python3. As Python2 is deprecated, and Python3 is a bit faster and more up to date.

I do not test the installation process that much, as it takes a lot of time. So basically, with a clean Buster running on a Pi4, you should be able to install the software with the following commands:

`sudo apt -y install git`
`cd /home/pi/`
`git clone --recursive --depth 1 https://github.com/theyosh/TerrariumPI.git`
`cd TerrariumPI`
`sudo ./install.sh 3`

And reboot. Make sure that you do this at a clean Raspian os. If you have installed my software before, there will be some changes in configurations that will break it when you try to reinstall it with Python 3.

cyber-shifter, 2020-04-19:
Ok, so. Everything seemed to go ok. Looks like it installed. Got the TerrariumPI splash screen when I SSH'd in. I went to the interface in a browser, and went to settings to change the password. Put in the defaults, it didn't take it. I can't set the admin name without putting in the credentials. The "pi" user credentials didn't work either. Is there a way I can reset the password through the CLI?

cyber-shifter, 2020-04-19:
So, I did a complete wipe and reinstall. Figured maybe having a special character in my admin account name might have fudged it. So, finally got back to the interface after reinstalling. Went to the settings page. Changed the admin name and password. My password has special characters, but the name does not. Hit submit. It gave me the green little changes saved message. Then, I changed other settings like the URL for the weather, measurement defaults, etc. Then I went to save again. It prompted me for a password. None of the known passwords work. Not the default, not the "pi" user one, not the new one. This is the second time this has happened consistently. This is a new Raspberri Pi 4.

cyber-shifter, 2020-04-19:
I should note, I have the same issue with languages now that I had before. I have no language choices, and cannot save settings changes because of this. Also, still getting prompted for a password, and none of them work. I have rebooted a number of times, the only thing that has changed is the IP address because I made a reservation.

cyber-shifter, 2020-04-20:
I did another reinstall. Did the same setup steps as before. Changed JUST the admin password. Saved. Then I went back out the Home page, and re-entered the setting page. I got prompted for my password, I entered the new one, and it worked! I have not yet tried changing the admin name yet, but I'm tired of fooling with reinstalls for the night, so I won't at this moment.

theyosh, 2020-04-28:
I think it should work. But I can also imagine that some changes will not work instantly and needs a restart. Should not be, but could be. But those issues are that rare, it is hard to debug.

If you want to check if you settings are saved correctly, look in the settings.cfg file. It is not hard to read and understand.

Because I would like to close this issue, as I cannot reproduce it. Any updates?

cyber-shifter, 2020-04-28:
I would be happy to check the config file and make sure the changes are there. Should I try changing the admin username to see if the change sticks?

theyosh, 2020-05-01:
Hi, you could do that. You are free to do so. But I think it will work, but this is such a change, that maybe a restart is required.

So if you are going to test this, you should look at the settings.cfg file and see if the values changes when you save it in the web interface.

theyosh, 2020-05-20:
Admin user is changeable. It will be stored in the settings file. So it is working like expected. Closing issue.

closed 2020-05-20 by theyosh