theyosh/TerrariumPI #378: Weird CPU Temp Status?

cyber-shifter, 2020-04-20:
So, since I live in the US, I'm more familiar with Fahrenheit for temperature. So I switched it in settings for the temperature indicator to be that way as well. Then I went to the system status page, to just see what the load on the PI was, with nothing running. There was a temp warning for the CPU, at around 110 degrees or so. I thought this odd, since it didn't feel hot, and it wasn't actually doing anything. Just out of curiosity, I switch the temp setting back to Celsius. I go back to the status page, and it gives me 42.35 degrees Celsius, with no warning. It still has the "F" for Fahrenheit next to it, but I figure that is a typo. So the conversion is right, but when on Celsius, the needle is in the middle of the gauge, and pegged to the right when the setting is on Fahrenheit. Not sure if this is a known, if it is, then disregard.

theyosh, 2020-04-21:
As I understand correctly, the value is correct in F, but the scale of the gauge is wrong. That can be fixed. Also, when you change from F to C or back, you need to refresh the hole webpage. Else it will not update the temperature indicators

theyosh, 2020-04-21:
Hmm, code does take care of it. So it should work. Make sure you have refreshed the webpage. And / or down the shift key when reloading

cyber-shifter, 2020-04-28:
Odd. I booted it up today, because I finally got my sensors in so I wanted to play with it. Didn't see the same issue. Thanks for giving me the info though!

closed 2020-04-28 by cyber-shifter