theyosh/TerrariumPI #379: Hardware Support for TP-LINK Kasa series of Smart Switches (HS-300, KP-303 etc)

bendobot, 2020-04-21:
Hey thanks so much for all your hard work on this project! I'm still pretty new to it, so apologies if this is possible to accomplish another way. Hopefully I can contribute somehow even if it's just testing.

Was wondering if you could support the TP- Link Kasa series of smart switches(HS-300, KP-303 etc) if you ever have time to add it? I've stumbled here from the reef-pi project which is go based with node UI and they have a driver for them here:
https://github.com/reef-pi/drivers/tree/master/tplink

It seems to connect to port 9999 via the IP to control it. Let me know if there's anything I can do to help with this...could provide a pcap or something?

Thanks again, your efforts are so much appreciated!

theyosh, 2020-04-21:
I guess so: https://github.com/python-kasa/python-kasa :)

I need to see how it works, but looks like those can be supported

bendobot, 2020-04-22:
Oh wow, you're awesome. There's a lot more devices in that list than I realized that can be supported too, cool! 

Whenever you get around to committing I will test a HS300 at least.

theyosh, 2020-04-28:
Hmm, looking a bit more closely, it looks a bit harder. You have a powerstrip with multiple sockets? That sucks :)

A single socket is easy. The strip is a bit harder to implement... But I will give it still a try.

bendobot, 2020-04-28:
ya I have the one with six sockets. sorry to be a pain :). Its rated very well online, and seems to fit the use case very well (where else other than terrarium automation do you need a six socket smart strip anyways?). 

I've originally built a 4-gang receptacle A/C box controlled by an 8-channel relay as well but it takes up too much space in the project box and too many gpio ports. the meross one stores password plain text, and the other ones we can't get here in North America (wemo is good but I don't have it!). I was thinking of getting one of the three port strips to control as well if it works. 

I'll send you some eth or xmr to buy hardware (or a beer) for testing :). Let me know if there's anything else I can do to help (provide logs or anything).

bendobot, 2020-05-06:
Hey noticed you committed something here, thanks again. 

In case it's not too late, I did find these:
https://www.reef2reef.com/threads/reef-pi-an-opensource-reef-tank-controller-based-on-raspberry-pi.289256/post-7375923
https://www.softscheck.com/en/reverse-engineering-tp-link-hs110/

Hope they help a bit.

theyosh, 2020-05-07:
Yes,

it took some time to figure out how the kasa library was created and how to use. It needs at lease python 3.5 So make sure you are using Raspian buster with a Python3 installation. That should be enough to get this Kasa support running.

For now, I have made a discovery method which should find all your power switches. But I known that it will crash when it is done. I cannot test that. I need first the feedback of this part of the code.

So, are you able to just use a second Pi, and install my sofware on that. So we can do some tests. We do not need sensors or other stuff. Just a clean install, where we can test the powerswitches.

That means that you have to do a full clone from this repository. And check out the branch feature/379

1. `git clone --recursive https://github.com/theyosh/TerrariumPI.git`
2. `cd TerrariumPI`
3. `git checkout issue/379`
4. `sudo ./install.sh 3`
5. `Reboot`

Then I need the information in the screen session. There will be more debugging going on. And I expect some errors.

If you are able to test this for me, and send the results back, I hope we can then support the power switches.

bendobot, 2020-05-07:
Hey I've checked issue/379 out now and am installing. Once I have the screen session up i'll let you know. 

I am testing now on a pi 3, but once this is done I can also test on a pi zero w (and even a pi 4 if you want). I will also be getting the kp-303 (it is currently on back order) which is the 3 port variant and can test against that if it helps.

bendobot, 2020-05-07:
It's up and screen is attached. Is this what you're looking for?

`Start scanning.....
[]
Found device
192.168.3.243
['__add__', '__class__', '__contains__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getitem__', '__getnewargs__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__mod__', '__mul__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__rmod__', '__rmul__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'capitalize', 'casefold', 'center', 'count', 'encode', 'endswith', 'expandtabs', 'find', 'format', 'format_map', 'index', 'isalnum', 'isalpha', 'isascii', 'isdecimal', 'isdigit', 'isidentifier', 'islower', 'isnumeric', 'isprintable', 'isspace', 'istitle', 'isupper', 'join', 'ljust', 'lower', 'lstrip', 'maketrans', 'partition', 'replace', 'rfind', 'rindex', 'rjust', 'rpartition', 'rsplit', 'rstrip', 'split', 'splitlines', 'startswith', 'strip', 'swapcase', 'title', 'translate', 'upper', 'zfill']
`

theyosh, 2020-05-07:
Hi,

yes and no :P I was hoping on something different, but is there not more output. I do expect some errors. So is there some more output?

bendobot, 2020-05-07:
this is all of it:

`2020-05-07 16:49:52,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.3.152:8090 ...
2020-05-07 16:50:20,975 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2020-05-07 16:50:26,032 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Start scanning.....
[]
Found device
192.168.3.243
['__add__', '__class__', '__contains__', '__delattr__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getitem__', '__getnewargs__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__mod__', '__mul__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__rmod__', '__rmul__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', 'capitalize', 'casefold', 'center', 'count', 'encode', 'endswith', 'expandtabs', 'find', 'format', 'format_map', 'index', 'isalnum', 'isalpha', 'isascii', 'isdecimal', 'isdigit', 'isidentifier', 'islower', 'isnumeric', 'isprintable', 'isspace', 'istitle', 'isupper', 'join', 'ljust', 'lower', 'lstrip', 'maketrans', 'partition', 'replace', 'rfind', 'rindex', 'rjust', 'rpartition', 'rsplit', 'rstrip', 'split', 'splitlines', 'startswith', 'strip', 'swapcase', 'title', 'translate', 'upper', 'zfill']
2020-05-07 16:50:41,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
`

bendobot, 2020-05-07:
192.168.3.243 is the correct ip address for the device.

bendobot, 2020-05-08:
Hmmm I reinstalled everything from scratch and started again. Nothing in the screen other than what was output. 

I tried to enable debug logging in the logging.custom.cfg by adding the fileHandlerDebug to the handlers in logger_root, but still nothing. I also enabled syslog logging. Is there a different logging facility that can be enabled for kasa?

It shows up in the switch settings in the frontend as tplinkkasa but I cannot select it in the pull down menu.

theyosh, 2020-05-08:
At the moment, I am only working at the auto discovery of the power switches. So the interface is not 100% done. First I would like to see if I get auto discovery to work.

And then I have to see how we continue. And the main issue right now, is that the library is written in Python 3.5 code, which has some funky new features that are being used. So for me that means a bit different way of thinking and code making.

I made an update, so could you do a `git pull` and just restart? See if we get some more information.

bendobot, 2020-05-08:
Got you. That makes perfect sense.

Now we got something:
```
Start scanning.....
[]
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 27, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 147, in __init__
    self.__load_power_switches()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 304, in __load_power_switches
    for power_switch in terrariumPowerSwitch.scan_power_switches(self.toggle_power_switch,**self.config.get_meross_cloud()):
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 1258, in scan_power_switches
    switches = power_switch_device.scan_power_switches(callback, **kwargs)
  File "/home/pi/TerrariumPI/terrariumSwitchKasa.py", line 109, in scan_power_switches
    asyncio.run(scan())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/TerrariumPI/terrariumSwitchKasa.py", line 81, in scan
    for ip_address, device in devices:
ValueError: too many values to unpack (expected 2)
```

theyosh, 2020-05-08:
Nope, this is not helping.... but I made another change in code. Another `git pull` and restart.

Another option is that if you trust me enough, give SSH access to the test Pi at your place. This can be done with port forwarding in your router. If you think this is a solution, please set it up, and mail login details to github.com@theyosh.nl

bendobot, 2020-05-08:
Hmmm. I tried another git pull and the screen wouldn't come up. Let me try a few things. I can't quite give you access just yet, but give me a few days, I'll make a DMZ for you to SSH into, then send you the login details to your email.

bendobot, 2020-05-08:
```
2020-05-08 19:47:22,013 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Start scanning.....
[]
Found device
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 27, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 147, in __init__
    self.__load_power_switches()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 304, in __load_power_switches
    for power_switch in terrariumPowerSwitch.scan_power_switches(self.toggle_power_switch,**self.config.get_meross_cloud()):
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 1258, in scan_power_switches
    switches = power_switch_device.scan_power_switches(callback, **kwargs)
  File "/home/pi/TerrariumPI/terrariumSwitchKasa.py", line 110, in scan_power_switches
    asyncio.run(scan())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/TerrariumPI/terrariumSwitchKasa.py", line 84, in scan
    print(device)
  File "/usr/local/lib/python3.7/dist-packages/python_kasa-0.4.0.dev0-py3.7.egg/kasa/smartdevice.py", line 673, in __repr__
    self.model,
  File "/usr/local/lib/python3.7/dist-packages/python_kasa-0.4.0.dev0-py3.7.egg/kasa/smartdevice.py", line 110, in wrapped
    assert self._sys_info is not None
AssertionError
2020-05-08 19:47:39,000 - INFO    - terrariumWrapper     - Restart counter: 2/5
2020-05-08 19:47:39,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 32 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2020-05-08 19:47:49,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.3.152:8090 ...
```

bendobot, 2020-05-09:
I put the pi on a separate vlan and sent you the details to login. Check your inbox for the credentials.

theyosh, 2020-05-09:
Yes, got it working. So you could test it yourself. Also, I was not able to toggle the power switch with an external program. If you have an app on your phone for that, try that also to see if my software will pick up the changes from outside the software. Will take up to 30 sec.

bendobot, 2020-05-09:
So I can still toggle the switches with the app that comes with it, even
when on a separate network. I've turned on plugs 1, 3, 5 through the app,
and can toggle them through the web interface. Very nice!

theyosh, 2020-05-09:
Nice! I did not much testing, but I have seen that changing fast a button or multiple button, it can happen then one action fails. I know why, but I have no idea how to solve it. But it will only happen when you try to switch all the buttons within 1 a 2 seconds. So toggle slowly! :)

bendobot, 2020-05-09:
don't go crazy, got you. I'll try to test a bit and if there's anything major is it better to re-open this or open another new issue?

I was thinking if there are any other scenarios you would like to test while I have the vlan up as well...I have an atlas scientific ezo circuit ph probe (too expensive) and a dfrobot conductivity probe. Will be getting a lot more hardware in the near future. If you think it's good to open issues for those various pieces of hardware let me know!

theyosh, 2020-05-13:
Still working? Found already some strange things? If not, I am planning to merge this with the master branch.

About your other hardware, the PH Probe is something that we can use. It can use i2c. Which is the easiest way. The conductivity probe is analog, and needs extra hardware for the Pi. So that is something which I will not support.

If you want, it would be nice to setup the PH probe and try to support it. But I suggest to open a new issue for that.

bendobot, 2020-05-14:
I haven't been able to test as much as I wanted yet, but from what i did i
was very happy with it. Feel free to merge with the master branch, and if
there is other problems I can open a new issue.

As far as the other hardware, I'll open an issue for the pH probe then,
I've got the ezo circuit programmed for i2c protocol currently, but it can
support UART too (I think pi3 failed at their UART implementation?). Let's
keep it simple and stick to i2c. Would be nice to make it work for other
people and add more hardware, and our setup seemed to work well for testing.

Regarding the conductivity probe I've got an ADS1115 and a MCP3008 (rasp.io
Analog Zero) if you ever change your mind :).

If you need any help documenting the process for installing as well let me
know, I would be happy to help.

bendobot, 2020-05-16:
I tested some more tonight. My KP303 arrived (the three port variation). It also works really well with this. Both are working together on the same network with no issues so far.

theyosh, 2020-05-16:
Cool! I merged this branch. So for you, just return to the master branch and do a `git pull` and you should be up to date again.

Thanks for testing

closed 2020-05-16 by theyosh

theyosh, 2020-05-27:
Hi,

the Kasa library has changed a lot, so I was wondering if you are able to test the changes.

`git pull`

stop terrarium pi

`sudo ./install.sh 3`

start terriumpi

Also, if everything went correctly with the installer, you can run: `kasa --version` and should point to version 0.4.0 dev

bendobot, 2020-05-28:
Hi...I had to wipe the sdcard, so I re-installed from scratch. Followed
these instructions on github and installed with python3, so that's what's
on there right now.

It won't start up now though...I'll send you the ssh details to connect
again to the same email as before if you want.