theyosh/TerrariumPI #380: Advice on heaters

martynwheeler, 2020-04-21:
Hi,
My terrarium is currently heated with an incandescent lamp via a microclimate mini stat 100. If I use terrariumpi to set the heating environment with a temperature sensor it will switch the lamp on and off.  Is there a better lamp option that would dim rather than switch? I was thinking about taking the ministat apart to see how the temperature regulator worked. I'm guessing that the dial is a potentiometer the controls a pwm dimmer. Or do you think that just setting the temperature and using terrariumpi to switch on and off via timer is best.  I'm fairly new to keeping bearded dragons.

Thank you 

Martyn

theyosh, 2020-04-22:
If you are using this one: https://www.microclimate.co.uk/product/ministat-100/ you can ditch it! :)

My software does exactly the same thing. This device is just an on / off switcher. So there is no dimming in it. So basically, you can hook the light directly to a relays that is connected with the Pi. And then in the environment settings at heater, you can specify your sensors and the power switch which is controlling the heating light.

This will just switch on and off when needed. On when the temperature is below the min value(s) of the sensor(s) and power Off when the temperature is above the max value(s).

Using a dimmer, I have only experience with: https://www.tindie.com/products/nEXT_EVO1/universal-ac-mains-dimmer-mpdmv41/ You can just hook up a 220V light bulb to it, and it will dim.
Shipping is rather expensive but this device works like a charm.

I have not found a lot other 220V dimmers that could be controlled by a Pi. Maybe: https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html

martynwheeler, 2020-04-22:
Oops I misread the box, it is actually a microclimate Dimmer B1 HT.  It just gradually dims and warms to provide a stable temperature environment.  If I hook up the environment sensor it will switch it on and off.  Maybe best to stick with a timer.

martynwheeler, 2020-04-22:
I see they have an updated model, MPDMv7.5.  Would recommend staying away from that?

martynwheeler, 2020-04-24:
Hi Again,
Sorry for all the questions.  I got hold of next_evo dimmer.  very easy to install thanks for the recommendation.

I am just trying to understand the settings now.  I have it all hooked up and it is working.

Here is the setting I have chosen for the switch. I was not sure what sort of values to go for.
[image]

My unserstanding is that this will change the value by 4% and hold it for 10 seconds before changing again?

In my sensors I have:
[image]

Does this mean that the dimmer will try and stay between 40 and 45 C

and the enviroment:
[image]

I am not clear what the power on is doing in this part?

Thanks again for your help and advice,

Martyn

cyber-shifter, 2020-04-28:
I too would like to know about this. I am planning to use radiant heat panels for my enclosures. I have read a bunch of places that it is better to dim the power to the panel to avoid the temperature "slop" that happens with just switching on and off. Being new to controlled systems, is it really worth being concerned with this "slop". I will be keeping snakes within my enclosures.

theyosh, 2020-04-28:
> Hi Again,
> Sorry for all the questions. I got hold of next_evo dimmer. very easy to install thanks for the recommendation.
>
> I am just trying to understand the settings now. I have it all hooked up [...]
Almost. This will tell the dimmer to change 4% in a duration of 10 seconds when the environment give the command to power up. Down goes the same way.
So, a single up or down action will take 10 seconds and will increase/decrease by 4%.

> 
> In my sensors I have:
> [image]
> 
> Does this mean that the dimmer will try and stay between 40 and 45 C
No, not at this point. At this point, you have only configured the sensor. So the limits are here at this point, only indicators. But they will be used in the next step in the environment. And yes, this will then tell to stay within 40 - 45 degrees.
> 
> and the enviroment:
> [image]
> 
> I am not clear what the power on is doing in this part?
Here, you have selected sensor mode with the correct sensor so that is correct. And I expect you only have a heater. So in the 'low alarm' you select the dimmer.
- The value 'power on' should be zero. This field will tell what the max period is to put the switch on. Select 0 and it will stay on for ever. Or in your case until it gets to hot.
- The value 'Settle timeout' is the amount of time to settle the sensors to the new temperatures. So I would select a value between 60 - 120 (which are in seconds).

Other values you have selected can work. But you have instructed/setup that the temperature at night should be 20 degrees lower. Is this correct? That means the temperature needs to be at 20-25 during the night? If so, it could happen that this dimmer is also working at night. If the heat is a light, it will never be dark.

As you have selected a dimmer, and at the dimmer there is 4% change in 10 seconds. After that change, it waits for the settle timeout to check if the dimmer needs more or less power.
When the max alarm is hit, the power will go down.  Until it gets below the low alarm value. Then it will turn up the dimmer until it is above the max alarm value.
Unfortunately I could not make this in such a way that it will just stay in the middle with the power values. So the dimmer will go up and down all the time. But as it is a dimmer, it should not be that disruptive for the reptile.

So with this setup, you can only heat your terrarium. There is no cooling option. For that you need another device on the low alarm part if you want.

I hope this will clear it up.

martynwheeler, 2020-04-28:
That is very helpful.  Thank you very much

closed 2020-04-28 by martynwheeler