theyosh/TerrariumPI #381: Water flow sensor

cyber-shifter, 2020-04-28:
This might need to be a "wish" but unsure. 

I got a Digiten Water Flow Sensor model FL-S402B. It was cheap enough, so I figured I could play with it. The idea is to put a flow sensor before my sediment filter on the misting system, so I can know how much water has gone through the filter (as they only have a certain life expectancy of 3600 gallons I think). The wiring is VCC+, GND, and Signal coming off the sensor (red, black, yellow) but I'm not sure exactly how I could hook it to Terrarium PI, or even if it would work. I would want it to count the flow cumulatively, not just sense the current status. So I figured I would just start by asking the question. I would be happy to help develop this feature if it can't already be done.

theyosh, 2020-04-28:
Hmm, that will be hard to make for me. But if you are handy, you could make it your self:

Here is some code example: https://raspberrypi.stackexchange.com/questions/34480/how-to-use-the-water-flow-sensor-with-raspberry

The issue is, that you have to count pulses in order to determen the current flow. I had issues with a simulair code part, so I had to change it. And that means that I cannot work with callback events as in the example.

Second issue is that you want kind of a countdown till it is 'empty'. That means I need to know how much there is, and how much water can flow. That does not fit in my current setup. I do have a watertank option based on a height sensor. But that works different.

So as an alternative, you could make your own script that listing at the flow. That script knows the total amount of the tank, and the water flow values. So that script can then count down when the water is flowing. And when that is going on, you can also write a JSON file to the static/extern folder, so that you can add an external sensor that measure volumes. In the JSON file you write down the current amount of water left. That will then be read by my software, and graphed. 
https://github.com/theyosh/TerrariumPI/wiki/Remote-data

I hope I was clear how you could implement it with a bit of custom code.

cyber-shifter, 2020-05-03:
I'm going to look into this. It's been a busy week in my world, so I'm just getting to this. Just to clarify, I don't want to count down how much water till empty, I just want to know how much has flowed through. I figured it could just count until I tell the counter to reset. I will look at it more. What is the current water flow on the dashboard tied into?

cyber-shifter, 2020-05-06:
I found what it was tied to. I'm looking into the code for the sensor now. I am going to play with the python code first separate of TerrariumPI. If I get code to work with it from a raspberry pi, could it be integrated as a sensor? I'm not sure what your procedure is for this.

cyber-shifter, 2020-05-07:
So that code works, and basically just counts up from initialization at zero. So at a basic level (because I don't know how to put it technically) I could store the count value, and just increment it whenever water flows through the system. I'm not a pi developer, but I'm guessing that a class could be made for the sensor that has it's specific flow calculation, and then the program could just take the flow number and do the math, if you selected that that was the sensor you were using. I can get my head around the programming, but do I have the basic gist right?

cyber-shifter, 2020-05-07:
I know I'm typing a lot here, and I apologize. I just wanted to put what I found before I sleep. The code that I used to get the sensor to work basically uses a lot of the same stuff I see in the terrariumGPIOsensor.py file. The one I have is simply a GPIO sensor. Just wanted to add that.

theyosh, 2020-05-07:
Hi sorry for the late reply.

I am still thinking of this, but I am afraid that it is not that simple. The main issue is, that the sensor is sending data when water is flowing. And that can theoretical happen without a power switch change. So in my code, that means that I have to have a process listen for data from the sensor.
I do this with the door, that is correct, but, that is done in not a nice way, which means, I look 10 times a sec for changes. For checking if a door is open or closed, this interval is enough. It is impossible to open and close a door withing 1 sec. But for a water sensor I am not sure.

Then, what do you measure with this. The door is only saying open or closed. There is nothing more. With the water sensor, you have also the value of how much is flowing. That means that a single sensor could have two functions.
 1. There is water flowing or not
 2. How much water is there flowing. This can be variable due to the source pressure.

So, that means that I have to make a graph, that shows the amount of water that is flowing. That graph will be updated with new data, once every 30 seconds. Which is enough for sensors. But for the water flow sensor, that is not handy. Because if the water is flowing for like 20sec, I could totally miss that in the graph, if I only check once every 30 seconds.

The current water flow is calculated based on the power switch duration. This is just done based of water flowing per minute or second. And I calculate based on the power switch changes. Using this logic, makes it easier to calculate the water flow. But there is no actual measurement.

So, the difference is that the water flow sensor is constantly sending updates when water is flowing, which all needs to be captured. Where the door and power switches only updates when the state is changing. So that is very less updates, and very more easy to handle.

Then, when we use this sensor as a normal sensor, and I will get data only once every 30 seconds, then how do we interpreter that data. Does that mean I have to divide the current value by 30, so we get a flow of liter/sec ?
Or will this be done on the same way as the doors, and I pull 1 times per second for data. And then show the value. Which means I get data graph points for every second. But, the code is currently only storing a value once per minute (if we use it as a sensor). Because else the database will grow that fast, that it will not work.

So, as you see, there is a lot more the do with this water sensor and the way you want to use. And you have your wish with this sensor, where another would like to have a graph that shows a total water level, and the sensor will see when there is water flowing, and will lower the total. That could also be a valid option for the use of this sensor. There is no singe way of using this sensor

cyber-shifter, 2020-05-07:
I have more to say later, but to clarify, I do plan to use this is conjunction with a switch. The switch would turn on to turn on the misting pump, and the flow sensor would be for the water going through the pump. I don't need the sensor to monitor at any other time, because the water is not moving except when the pump is on.

theyosh, 2020-05-07:
Ok, I am afraid that is just the thing that is not supported, and not easy to implement. So maybe we should discuss a bit more before someone is making a lot of code.

The current water flow is based on the power on duration and the entered flow value at the power switch. And this is hard to change.

cyber-shifter, 2020-05-08:
Could we leave the existing one as is, and just add a new one? One that is just a counter? It doesn't need to display the live flow, just a running total of what has run through so far. It could keep adding until you hit the reset button. Call it "filter life" or something? The code I am running the sensor with at the moment, all it does is add 1 to the count for each pulse. A simple bit of code would reset the count variable, and the flow sensor model would determine the flow equation. The model would have to be a supported one, so we could have the flow equation in the system, but that could be a drop down selection thing, to choose a supported model.

I am happy to write code that helps this, and test it. I admit I don't know this system super well, so I would need help as far as integrating it into TerrariumPI.

theyosh, 2020-06-01:
Sorry for the delay. There were some other issues more demanding.

What you have described here above, is your way of using a flow sensor. But other people would maybe want to use it differently. So my approach on this is, that a sensor should only provide data. No logic. The logic is then created by combining the sensor with switches in the environment part.

Also, adding somewhere in the interface the option to reset values or sensors is not a nice way of working. So I still think the only way of using this, is that the sensor is entered as a sensor. Where you enter what the amount of flow is per pulse. Because that is mostly specified.

And I think you should have some threshold value that is specifying if the flow is stopped. For example if there is nu pulse withing 2 seconds, the flow has stopped. This depends on how precise the flow server is. I have not seen this info yet, but this can be done based on some testing.

My only issue right now is that I cannot test it, and as this is in my opinion an active sensor, it is a bit harder to program. The sensor has to be able to make callbacks that the water is flowing rather then just asking what is the current state.

cmonicob, 2022-02-25:
Hello,

I think it could be done through the implementation of the arduino sensor, when doing all the calculations through the MCU, it could incorporate a pulse counter and simply check the pulse counter, you can also put an average of pulses per minute, and any other calculation that we need .
The good thing is that we only need to read the calculations made by the MCU.

At the moment the sending of data to the MCU is not implemented but we could expand it to be able to send orders such as resetting the counters.