theyosh/TerrariumPI #382: 2x USB Denkovi V1 4 relay board

bedrich19, 2020-04-29:
Hi,

I try to connect 2 x USB Denkovi V1 4 relay boards through FTDI, but only one  is working. Is it possible  to use 2 x usb relay board?

Thanks

theyosh, 2020-05-01:
Hi @bedrich19 

At the moment I only support 1 board. So that needs to be changed in my code. But I think it is not that easy. So that could take some time.

But, if you are handy, you could also create custom power switch scripts for each port on the second relay board. And them then as scripted power switches.

It is an old relay board that is configured as a FDTI switch?

Because the 'problem' could be that I have to change my code in such a way, that you cannot use the FDTI relais with other software. At the moment, I am only making connection when I need to make an action. But I am not able to directly select a FDTI relay board. So I have to loop over the devices in the hoop that the order is always the same. Also after reboots the order of devices can be different. So I need to figure that out. Also, what happens if you change the order on the USB connectors. You swap the devices over USB ports.... then I still need somehow to know which device is which... So that is the big question now.

I will give it a try

theyosh, 2020-05-01:
Hmm, apparently it is much easier!!!!

So now you can specify a second or more board by adding ',[board_nr]' to the power switch. So the full address is now: [switch numer],[board_number] where the board_number is optional. So existing users will not be affected by this.

So do a `git pull` and restart and you should be able to use a second relay board.

bedrich19, 2020-05-03:
Hi, you are the king. It works :)

Thanks a lot.

closed 2020-05-03 by bedrich19