theyosh/TerrariumPI #384: CPU overload

jeff26300, 2020-05-02:
Hello,

My CPU is constantly overloaded and has a permanent warning. Is it normal?

[image]

Thanks,

Jeff

jaysalazar, 2020-05-03:
Do you happend to have a CPU fan installed? I have a Pi3 w/nofan and Pi4 w/fan in a warm garage and the Pi4 w/fan helps alot.

theyosh, 2020-05-03:
Hmm, is it really overloaded... I think it is something different.

Are you using 1wire devices through GPIO? It not, you can disable it by changing the file: /boot/config:

`sudo nano /boot/config'

And add a '#' for the line 'dtoverlay=w1-gpio'

Then reboot it. This will stop polling for 1Wire. I have seen that it could increase the load a bit.

Else you have to look with top command to see what is taking a lot of CPU: https://www.lifewire.com/linux-top-command-2201163

And about temperatures, my Pi 4 is running at 60C which is fine for me. My Pi3 is doing 55C So that is fine for me, and do not see that as a problem.

So could you look with top command to see what is using a lot of CPU?

jeff26300, 2020-05-05:
Thanks @theyosh , I have indeed a 1wire for the hearing sensor. I run the command you suggested and this is what I got: 
[image]

@jaysalazar , I have currently no fan on my pi zero

Thanks guys!

theyosh, 2020-05-05:
Ok, then the 1wire is not causing this.
Then I see two python processes running. And my software should only have one python process. So what you can do now, based on the information of top. Run the following command:

`ps fax | grep 1823` and see what the output is. The number 1823 is the number in the first column on the image above. That is also the process which takes the most cpu.

So the then you can what both python process are. Repeat the above for all the python processes.

jeff26300, 2020-05-05:
that's what I got for the main one :
[image]

theyosh, 2020-05-07:
Hmm, you last picture is not correct. I need the line where it starts with that number 3776. Now you have only listed the process 'grep' which is not the right line.

A total different option can be:

1. Do a `git pull`
2. Rerun the installer as you have done with the normal installation
3. Reboot

Then we now for sure you have the latest software with the latest libraries. If that is solving the issue, then it will be hard to figure out how it started.

jeff26300, 2020-05-10:
At the end we decided to reinstall the system and everything is now working fine :)
Thanks for your support! I close this issue

closed 2020-05-10 by jeff26300