theyosh/TerrariumPI #385: Environtment Settings Not Turning Off Switch

jaysalazar, 2020-05-03:
Hello, first I want to say how awesome TerrariumPI is, great job to theyosh for all your hard work and ongoing support.

I have an issue that may be configuration related so help from anyone would be much appreciated.

I setup my temp sensor with Alarm Min = 70F and Alarm Max = 78F, I configured my Environment settings to turn on a switch (connected to cool air duct) to come on when temps go over 78F which actaully works. The problem is that when the temp comes down below 78F the switch never turns off. I have attached my settings, the switches are not in "overide mode" and I have tried restarting.

Any ideas?

[image]
[image]

theyosh, 2020-05-03:
It will shutdown when it gets below the min value. So when it get to hot, it will cool down until the min value. Then it stops, and waits till the temperature is above the max value. 

So if you want to stay the temperature about 78F, you should use 78F as max, and 77F or 76F as min value.

jaysalazar, 2020-05-03:
I adusted the min value and it's working as designed, thank you for your prompt response! 

Quick question, I get the following warnings, I think they have been addressed here in the past. I assume they are safe to ignore?

[image]

theyosh, 2020-05-05:
Those messages are there, because the change in value is more then the allowed change in value in 1 measurement. Per sensor you can give a value what is the max difference per measurement.

Some sensors are not that stable and can sometimes give strange values. So when the change is more then the entered value at the sensor, the measurement is ignored. But after 5 times, it looks like the new measured values are correct and will be used.

This can happen when you turn on a heater which is capable of heating up pretty fast.

So if the measured values are somewhat valid to the conditions in your terrarium. you can just ignore them. If the values are not correct, and the sensor is really broken, a hardware replacement is needed.

jaysalazar, 2020-05-06:
Thank you for the explanation, I will close this issue now.

closed 2020-05-06 by jaysalazar