theyosh/TerrariumPI #387: Temperature control issue, conflicting switch status

jeff26300, 2020-05-05:
Hello,
I am back again on a temperature control issue. I still have the same set up as described in issue #344 and usually everything works fine. But time to time the heater is not controlled by the sensor and exceeds the alarm value and never stops (see picture). Some days it goes very well and other days we have to stop the heating in order not to get fried frogs... What we found also strange is that time to time the button of the heater switch appears blue (off) and its graph shows power running, which is true because the temperature increases. Can it be related to a full history log that I need to purge? Or the CPU overload described in issue #384 ?

[image]

Cheers

jaysalazar, 2020-05-07:
What type of temp sensor are you using? Can you show picture of your temp enviroment settings?

theyosh, 2020-05-07:
@jeff26300 Are you using a switch or a dimmer?

And on the graph, the temp goes down at around 10:30. Is that due to a manual power off of the switch?

And as @jaysalazar asks, what are your environment settings. Could it be that there is a rather high settle timeout? Because after the toggle action, it will wait for the settle timeout for it will do anything further. So if you have a rather high settle timeout, that could cause this.

jeff26300, 2020-05-08:
We indeed use a switch but we stopped the heater manually at that time, to save the frogs
We have a 1-wire sensor and the following environment :
[image]

This behaviour seems to be completely random

theyosh, 2020-05-08:
You should enter these values at the HIGH alarm settings. Not the Low. You want to control a heater, so use the High alarm values.

Clear the Low alarm values (at least the power switches)

jeff26300, 2020-05-15:
So we have tried many things and we discovered a bad electrical contact with the sensor :s  
After repairing it, we moves the values to the High alarm settings but the system was not working, it was switching the heater when the high temperature alarm was reached (which first sounded logical to me because I have always used low alarm to switch on the heater). We finally moved the values back to low alarm and it now works perfectly!! Would you want me to close the issue or you need to investigate further if the low and high settings are not reversed?

Thanks

theyosh, 2020-05-15:
Damm, I think I was not awake... You are correct. you need to use the LOW alarm settings. What was I thinking..... :(

But nice that you have found the issue. Hardware problems are hard to debug.... So if it is a hardware bug, and it is fixed now, you can close it.

Hardware issues I cannot debug, so there is not much to investigate anymore.

theyosh, 2020-05-17:
Closing now, as hardware issues are hard to debug. Nice that it is working now.

closed 2020-05-17 by theyosh