theyosh/TerrariumPI #388: Error on humidity and temperature sensor

itsmetrist67, 2020-05-07:
python error getting new data from dht22 centre error cannot inport name, rasberry__2_driver from adafruit_dht'(/usr/local/....

Please assist

jaysalazar, 2020-05-09:
I am going to assume you are using Python 3, first make sure its working outside of TerrariumPI.
1.Make sure Adafruit is installed
git clone https://github.com/adafruit/Adafruit_Python_DHT.git
cd Adafruit_Python_DHT
sudo apt-get install build-essential python-dev
sudo python3 setup.py install
sudo pip3 install Adafruit_Python_DHT
2.Go Inside this directory (SSH or Terminal) /Adafruit_Pyhton_DHT/examples
3.Run the following sudo ./AdafruitDHT.py 22 GPIO#  **Note the GPIO# is the BCM# not the Physical#

This should give you a reading for the Temp (Celius) and Humidity. DHT22/AM2302 are not stable they stop working after days/hours. Look into better sensors.

theyosh, 2020-05-09:
@itsmetrist67 what is you Raspberry pi device? Is it a type 2? Not sure if that will work. I only have tested this software on Raspberry 3 and 4. And zero should work with some tricks.

Raspberry Pi 1 and 2 are untested if they will work. GPIO could be different.

theyosh, 2020-05-17:
Please have a look at: https://github.com/theyosh/TerrariumPI/issues/392#issuecomment-629765331

It turns out that the DHT library is outdated, and that we need to use a new version. That does require the use of Python3.

theyosh, 2020-05-21:
It should be fixed. Please do an update and reinstall:

`git pull`
`sudo ./install.sh 3`

I suggest to install everything with Python3 to be future proof

itsmetrist67, 2020-05-21:
Thank you so much appreciate the help

closed 2020-05-21 by itsmetrist67