theyosh/TerrariumPI #389: Temperature regulation

martynwheeler, 2020-05-08:
Hi there,
Thanks again for this great piece of code and your helpful advice.  I was wondering why you have a set of boundaries for temperature rather than aiming for a specific temperature?  Would it be more efficient to implement a PID controller system (e.g. https://pypi.org/project/simple-pid/)?  Then the user would just set the desired temperature and it should control itself?  it might help to give a more stable environment?  Just wondering really.

Thanks

Martyn

theyosh, 2020-05-09:
Hm, I am unfamiliar with PID controllers. If I google for it, I do not find anything that could be usefull to me? Maybe I am missing something. I only find complete hardware systems. But then is my first question: 

are those not just dimmers?

So when I started this, I had the requirement to keep temperature and humidity at a certain range. In order to do that, is the easiest way to specify min and max values. And the act on that. To high, switch something. To low, switch something else.

As dimmers are harder to program then just on and off switches, I made what I call the fridge way. So when it is to cold, put on the heater until to hot, then shut down and wait till it is cold again.
And yes, that is not smart of efficient. But did the job. Kept humidity in a certain range. And you can do this with some heating mat for the temperature as well.

Next I supported dimmers, as there was some simple hardware to use. With the dimmer I increased power at small steps. And when the temperature is not to cold anymore, it will stop increasing the heat. And therefore, the dimmer will keep the heating at a stable temperature. And that sounds like the same thing a PID controller does. Because I already had the system in place for specifying the ranges, this dimmer update was just a small piece of code. All the logic could stay the same. So that was an very easy upgrade to my software.

Also, the project had as goal to use simple off the shell hardware. And I think PID controller hardware is not. That is a bit more complex and expensive.

Maybe I am missing something, but I do not understand how a PID controller system/hardware can be used with heating devices.

martynwheeler, 2020-05-09:
Hi there,

PID is not a hardware specific system, it is how the software controls/regulates the hardware.  It is basically as you describe: if too cold, heat a bit until it is too hot then cool.  What PID would do is to do this in a much more systematic way and keep the temperature locked to a much smaller range.  There is no need for any complex hardware, the dimmer module and a temperature sensor are all that is required (see this for example - https://onion.io/2bt-pid-control-python/).  I think that using this library (https://pypi.org/project/simple-pid/) it would be a few lines of code in your software.  The advantage would be much tighter control done in a mathematically systematic way and fewer parameters for the user to enter into the settings (they would just select the desired temperature, no need for max and min values and heater settle times as the control system would do this automatically).

I have a spare dimmer and temperature sensor.  I might have a look how easy it would be to set up and then I could report back with my findings.  What do you think?

Thanks, Martyn

theyosh, 2020-05-10:
Hmm, I see a bit of a different way of possible using this. Because I want to support multiple solutions with heaters, I will stick with the min and max values. That gives more flexibility then just 1 number.
But, then again, just take the average of the min and max value, and you have the desired temperature. But this is only valid for dimmer devices. I also have a heating map that works just on and off. And then I like to have a range to specify, rather then 1 average temp, and needs to switch on and off to often.
And this is only applying to the combination of a heater and temperature control. So, I was thinking, make it a setting for now. Which specify which logic you want to use for a dimmer for heating control.

Then in code, there could be different ways to achieve the temperature.

But, then I have seen the example code, and that means that my code needs to change a lot, because the software needs to be running all the time and I think read out the temperature sensor a lot more and often then once every 30 seconds.
And there is a part of dimming based on PWM which I am not that familiar with. So that was basically trial and error. And I am using http://abyz.me.uk/rpi/pigpio/pigpiod.html because that has a external daemon that controls the PWM signals. That makes it easier for me to use, then rather make my own loop that is continuously running creating PWM signals.

So, I see that this could be a nice addition, but at the moment I also see some issues. So if you have time and want to play with it, that is up to you. I cannot say right now if you make some changes that I can use them. I all depends how it will fit together.

martynwheeler, 2020-05-10:
Yes, I see now that this is maybe not as simple as I first thought for your code - i had not realised that it was not running continuously.  I might try and experiment anyway.

closed 2020-05-10 by martynwheeler

theyosh, 2020-05-10:
Yes, this was a design choice from me. Because if you add 10 dimmers, I get 10 running threads. Or I have to make some master thread which controls all the PWM. And when I found that external daemon solution, it was an easy choice for me.

Secondly, I have seen a lot of issues with some sensors when they are red every second. So that is way I only update every 30 seconds. That makes the sensors more stable, and I could work with that 30 seconds update system.

But I still did like this idea. So maybe in the future we can have multiple ways of controlling the dimmer