theyosh/TerrariumPI #390: Viewing temperature on Home Assistant

justin1977justin, 2020-05-09:
I have looked through all of the past issues in an attempt to resolve this myself but i am unable to figure it out. I have Home Assistant running and would like to be able to view one of my terrarium temperatures on it. I really have no idea how to begin. I have setup a sensor in home assistant trying to get cpu temperature as a start thinking that would be the easiest thing to setup. I used an example that i found for a reef controller and adjusted it to the following which is the only config that i have found that will actually show as a sensor in HA. However, the state just shows as {}. Has anyone incorporated this with HA and can point me in the right direction?

`  - platform: rest
      name: ClaytonGecko Temperature
      username: !secret GeckoUN
      password: !secret GeckoPW
      authentication: basic 
      json_attributes:
        - cpu_temperature
      resource: http://terrariumpi_ip_address:8090/api/info
      headers:
          Connection: keep-alive
          content-type: application/json`

martynwheeler, 2020-05-09:
Not sure that api/info is the right address? Can't see it in the wiki page https://github.com/theyosh/TerrariumPI/wiki/API

You are probably best just getting a sensor reading using GET /api/sensors/[sensorid].

theyosh, 2020-05-09:
Ha, that is explained here: https://github.com/theyosh/TerrariumPI/wiki/Integrations#home-assistant
I have not tested, but it should work

theyosh, 2020-05-09:
For CPU temperature it is a bit difficult. Because that information is at the API call: /api/system
But it will produce more info than just the CPU temperature. So I do not know if that will work.

But for a normal sensor the url should be what @martynwheeler suggest. /api/sensors/[id] And then use value templating: https://www.home-assistant.io/docs/configuration/templating/#processing-incoming-data

justin1977justin, 2020-05-10:
Thanks for the info. I did not find that that first link you listed when i was searching. I will try it. I am not very good with this stuff but trying to learn.

justin1977justin, 2020-05-12:
Just wanted to provide an update. I was able to get this working tonight. What you linked put me in the right direction but i could not figure it out until I did some more debugging tonight. First I was just attempting to get Home Assistant to show it as a sensor. I found that HA does not allow states values to exceed 255 characters. I added the value template that you list and it was not a valid configuration. I eventually found that it should be the following. All is working now. Thanks for your help.
`"{{ value_json['sensors'][0]['current'] }}"`

theyosh, 2020-05-12:
Nice. So it is working now for you? I made the Wiki information based on the documentation. I could not tested it. So that there was a small issue with quotes, that I could not see. So I have updated the Wiki to point that out.

So if it is all working now, can we close it? As you have re-opened it.

justin1977justin, 2020-05-12:
Yes. It is working now. I reopened to make sure you saw it. Closing it now.

closed 2020-05-12 by justin1977justin