theyosh/TerrariumPI #391: Water Mister

Hemibon, 2020-05-10:
Hello!
I am new to your TerrariumPi program but I am liking it a lot! I have a new RaspberryPi  4 that I am connecting to the terrarium I've built for my dartfrogs. My lights and heating lamps are already connected and working! Humidity/Temp sensor and distance sensor (for the water level in the tank) also!
I have a couple of questions about misting/spraying though:

1. I see in #231 hat you talked about a sprayer and timings of less than a minute. Did you finally manage to develop this?

2. I also see in Help, Environment, Sprayer (see attached pic) that there is something about this! But where exactly? I can't seem to find it in the System - Environment of my terrarium (I am using TerrariumPi 3.9.8).

What I would like to do is to switch my sprays for 10-20secs 2 or 3 times a day, independently of the humidity sensor. I've connected the sprays to the relay and they work when I trigger the relay.

Thanks in advance! 

[image]

theyosh, 2020-05-13:
Hi, yes and no :P The issue you refer to, is still on the wish list. But I think you can create a setup you want without that issue.

[image]

In the above image, I use the humidity environment to do almost the same thing. 
- Mode: timer
- Sensors: empty
- Day/night difference: what you need or want
- Day/night source: either use weather or lights

Then use the low part of the humidity settings.
- Start: Enter here the first time you want the sprayer to run. For your case you could take: 11 hours in the morning
- Stop: Enter the last time stamp of which the sprayer should/could be running. Lets say 2 hours before the lights go off. In your case: 6 o'clock in the evening. So the total duration of this timer is then 7 hours.
- Timer on: Here you enter 1. This means, that you want 1 minute to be powered on. This will be reduced to even less later on.
- Timer off: And now we have to calculate. Because the time entered here, will power down the power switch. We have 7 hours off total on time. Divide that by 3 (your amount of spraying during the day) Then you get about 2 hours and 20 minutes. So the on and off time together should cover 2 hours and 20 minutes. So here we enter the value 139 (minutes)
- Light state: I suggest on. So when the lights are off, it will never spray
- Door state: Ignore
- Power on: Here we can specify the on time for one action. And here you can specify seconds. So here you can enter the value of 10 or 20. Then  the sprayer will only be on for that amount of seconds.
- Settle timeout: Again, some calculation here. As the total on time is 1 minute, the settle timeout should be at least 60 - Power on duration from above. So in your case then 40 or 50 seconds. But to be sure, just specify 60 seconds.
- Power swiches: your power switch

So, this way, you have created a timer, that has 3 periods which are the Timer on + Timer off periods. And in that period, the power will be on max Timer on time. But with the second Power on and Settle timeout, we can control how long the sprayer is working withing that 1 minute. After that 1 minute, it waits the Timer off timeout, and start all over again.

I hope this is clear. And make sure you test it. So try it on a day that you are at home

theyosh, 2020-05-17:
Hi, was my explanation above clear? Do you need more information?

Hemibon, 2020-05-17:
Yes! It was very clear, and has been working very smoothly yesterday (4 times a day, 10sec each time at 8.00, 12.00, 16.00 and 20.00) and this morning. I wanted to confirm tonight's cycle for a full 48h run. I might as well close it now :) Bedankt voor je hulp!

closed 2020-05-17 by Hemibon

theyosh, 2020-05-17:
Nice, then if you have time please check: #210

Hemibon, 2020-05-17:
Yes :) I posted them a week ago. thanks again!

theyosh, 2021-10-30:
The SHT3XD should be working with the latest code.