theyosh/TerrariumPI #392: Fresh install

steveava, 2020-05-13:
It's been a while since I looked at TerrariumPi and now have some spare time to replace existing controls for our Dragons.

Still have the Pi3 and sensors laying around so thought I'd start afresh with a new install. Seem to be having trouble making it work and cant remember these issues previously so perhaps the install has changed, although it looks much the same from the instructions page.

I've tried Stretch Lite with updates/no updates/upgrades and get the same result. The service doesn't seem to start at boot and I cant get a page at port IP:8090. 

Tried this:

pi@raspberrypi:~/TerrariumPI $ sudo ./start.sh
2020-05-13 14:54:28,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server running as user 'pi' at location: http://10.132.40.203:8090 ...

No web service/page:

pi@raspberrypi:~/TerrariumPI $ sudo service terriumpi restart
Failed to restart terriumpi.service: Unit terriumpi.service not found.
pi@raspberrypi:~/TerrariumPI $

Seems to be a case difference between instructions and install. TerrariumPi vs TerrariumPI. Not sure if thats reflected elsewhere.

theyosh, 2020-05-13:
Hmm, at the moment I think it is the best thing to do is using Buster and go for Python3. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

As my old SD card died, I have to reinstall my Pi. And I did it with a Pi3 on Buster. And using Python3. That worked for me.

Stretch is older, and I do know that some python libraries are getting outdated. Also, after starting up, you should either look at the screen session or at the log files to see what is going on.

But I suggest to reinstall the OS with buster lite version. Desktop is not needed. And then install my software with Python3

steveava, 2020-05-14:
Thanks, seems to be working now. Did notice a few errors at boot. “Failed to start Backend server for 1-wire control” not that I am using any.

 

Tried to use a DHT22 but got this in the log. Perhaps DHT22 is a bit out dated and should be using something like Si7021. What do you think?

 

 

2020-05-14 12:12:53,313 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Humidity'. Error message: cannot import name 'Raspberry_Pi_2_Driver' from 'Adafruit_DHT' (/usr/local/lib/python3.7/dist-packages/Adafruit_DHT/__init__.py)

2020-05-14 12:12:53,199 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temperature' temperature from 25.62C to 25.61C in 1.01600 seconds

 

This email has been checked for viruses by AVG.
https://www.avg.com

steveava, 2020-05-14:
Didn't read the BME280 data sheet. Can get Humidity from that!

theyosh, 2020-05-16:
Hi,

a few things. Yes, BME280 does read out humidity. The difference with BME and BMP is air pressure. I use BME280 a lot. So that should work.

Then the DHT sensor. You have an error which somebody else also has. And the library that I am using the the DHT is a bit outdated. There is a replacement for it, but that means for Python3 only. In your case not an issue I see, as you are running with Python3.
So I will try to fix that.

BUT..... before you start your journey with the DHT22 sensor.... look here https://github.com/theyosh/TerrariumPI/issues?q=dht+is%3Aissue+is%3Aclosed ;) They are not stable for long runs. So I would suggest look for other humidity sensors, as for temperature the 1 Wire DS1820 are great! Or use BMD280

theyosh, 2020-05-16:
But, if you still have a DHT22 connected, are you able to do some testing for me? Then I will try to use a newer library for it, and that should fix the error 'Error message: cannot import name 'Raspberry_Pi_2_Driver' from '

justin1977justin, 2020-05-17:
I'm using a DHT11 on my original build which has been working great for a little over 1 week. I started another build with the same setup and I am getting i believe the error steveava mentioned. Only differences are the version of pi used and version of dht11 used.

theyosh, 2020-05-17:
Ok, I made a new branch for testing a new library for the DHT sensors. If you have time to test:

`git pull`
`git checkout new_dht`
`sudo pip3 install adafruit-circuitpython-dht`
`sudo ./start.sh 3`

The new library is Python3 only. So make sure you run the software with Python 3.

If you are not able to check the branch 'new_dht' then you need to do a complete new clone: `git clone --recursive https://github.com/theyosh/TerrariumPI.git`

steveava, 2020-05-17:
I’ll test it tomorrow morning.

 

https://www.avg.com

steveava, 2020-05-18:
Had a path error for new_dht so done a new clone.

 

Getting this error for DHT22 

 

2020-05-18 12:52:31,030 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'Teperature2' is outside valid range 15.00C - 45.00C in 0.00700 seconds.

2020-05-18 12:52:31,028 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Teperature2'. Error message: cannot import name 'Raspberry_Pi_2_Driver' from 'Adafruit_DHT' (/usr/local/lib/python3.7/dist-packages/Adafruit_DHT/__init__.py)

2020-05-18 12:52:30,916 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temperature' temperature from 27.66C to 27.66C in 1.01423 seconds

 

If it helps:

 

pi@raspberrypi:/usr/local/lib/python3.7/dist-packages/Adafruit_DHT $ ls -l

total 88

-rw-r--r-- 1 root staff  5779 Jul 25  2018 Beaglebone_Black.py

-rw-r--r-- 1 root staff  4324 Jul 25  2018 common.py

-rw-r--r-- 1 root staff  1176 Jul 25  2018 __init__.py

-rw-r--r-- 1 root staff  4494 Nov  7  2018 platform_detect.py

drwxr-sr-x 2 root staff  4096 May 14 10:43 __pycache__

-rw-r--r-- 1 root staff  1891 Jul 25  2018 Raspberry_Pi_2.py

-rwxr-xr-x 1 root staff 43000 May 14 10:43 Raspberry_Pi_Driver.cpython-37m-arm-linux-gnueabihf.so

-rw-r--r-- 1 root staff  1984 Jul 25  2018 Raspberry_Pi.py

-rw-r--r-- 1 root staff  1624 Jul 25  2018 Test.py

 

--------------------------

 

 

I’ll be using the bme280 anyway but let me know if you want me to test the DHT22.

 

https://www.avg.com

theyosh, 2020-05-18:
I need the error when on the new_dht branch. The current DHT library is broken and will not be fixed.

So we have to test the new_dht branch

Did you install the new library: ` sudo pip3 install adafruit-circuitpython-dht` ? That is needed

steveava, 2020-05-18:
got this error from git checkout:

pi@raspberrypi:~/TerrariumPI $ git checkout new_dht
error: pathspec 'new_dht' did not match any file(s) known to git

Then went for the second clone option.:

git clone --recursive https://github.com/theyosh/TerrariumPI.git
sudo ./install.sh 3
sudo pip3 install adafruit-circuitpython-dht

or should the new library go in before install.

theyosh, 2020-05-18:
Oh, my mistake. I thought you had some git experience.

After you have done: `git clone --recursive https://github.com/theyosh/TerrariumPI.git`

then do `git checkout new_dht`

and then it should be working. I assume you have already run the installer once, and you have already did ` sudo pip3 install adafruit-circuitpython-dht`

If not sue, you can always rerun the installer: `sudo ./install.sh 3`

steveava, 2020-05-18:
No, dont use git much but the order you have is pretty much what I went through.

I'll start again to be sure.

git clone --recursive https://github.com/theyosh/TerrariumPI.git

This is fine.

git checkout new_dht

I get:

pi@raspberrypi:~ $ git checkout new_dht
fatal: not a git repository (or any of the parent directories): .git

or from /TerrariumPI:

pi@raspberrypi:~/TerrariumPI $ git checkout new_dht
error: pathspec 'new_dht' did not match any file(s) known to git

Then went for "sudo pip3 install adafruit-circuitpython-dht"

Re-run "sudo ./install.sh 3"

Tried "sudo pip3 install adafruit-circuitpython-dht" just to be sure but says it already satisfied.

Then went for "git checkout new_dht" and get the same errors above.

Not sure what I am doing wrong.

theyosh, 2020-05-18:
Hmm, again I missed a small step.

So, you have done now a full git pull. So you should have a folder called 'TerrariumPI'

`cd TerrariumPI`
`git checkout new_dht`

That should do the trick. Lets try that. The git checkout command needs to be run in the folder TerrariumPI

steveava, 2020-05-18:
Yep, full git pull but this is what I get from “git checkout new_dht” while in /TerrariumPI

 

pi@raspberrypi:~/TerrariumPI $ git checkout new_dht

error: pathspec 'new_dht' did not match any file(s) known to git

 

https://www.avg.com

theyosh, 2020-05-18:
I don't get it. So we are doing it in a different way.

Go to the folder TerrariumPI in the pi home folder.

`cd /home/pi/TerrariumPI`
`rm terrariumGPIOSensor.py`
`wget https://raw.githubusercontent.com/theyosh/TerrariumPI/new_dht/terrariumGPIOSensor.py`

And then just run: `python3 terrariumPI.py` So that it will just stop when there are errors. That makes it a bit easier to debug.

steveava, 2020-05-18:
Getting a different DHT22 related error now.

 

----------------------------------------

 

pi@raspberrypi:~/TerrariumPI $ python3 terrariumPI.py

2020-05-18 17:51:05,456 - WARNING - terrariumSensor      - OWFS file system is n                                                                                                                                                              ot actve / installed on this device! If this is not correct, try 'i2cdetect -y 1                                                                                                                                                              ' to see if device is connected.

2020-05-18 17:51:10,499 - WARNING - terrariumBluetoothSensor - Bluetooth scannin                                                                                                                                                              g is not enabled for normal users or there are zero Bluetooth LE devices availab                                                                                                                                                              le.... bluetooth is disabled!

2020-05-18 17:51:15,566 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Blue                                                                                                      tooth LE devices available.... bluetooth is disabled!

/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: error while loading shared libr                                                                                                      aries: libgpiod.so.2: cannot open shared object file: No such file or directory

2020-05-18 17:51:15,871 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'temp'. Error message: Timed out wai                                                                                                      ting for PulseIn message. Make sure libgpiod is installed.

2020-05-18 17:51:15,874 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'temp' is outside valid ra                                                                                                      nge 0.00C - 100.00C in 0.29892 seconds.

2020-05-18 17:51:26,000 - INFO    - terrariumWebserver   - Running webserver at :::8090

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

justin1977justin, 2020-05-18:
I did a clean install then installed library and checkout worked. I am
working on making my case so I dont have the sensor connected at the
moment. I will try to connect it soon and get back with you.

theyosh, 2020-05-18:
@steveava Ah, then the next step:

run: `sudo apt -y install libgpiod2`

And then restart TerrariumPI with: `python3 terrariumPI.py`

steveava, 2020-05-18:
Ok thats installed. Similar error except for the invalid GPIO......

 

2020-05-18 18:26:51,590 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!

invalid GPIO offset: False2020-05-18 18:26:51,894 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'temp'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-18 18:26:51,897 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'temp' is outside valid range 0.00C - 100.00C in 0 in 0

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

steveava, 2020-05-18:
Missed a bit at the bottom of the page.

invalid GPIO offset: Falsesys:1: ResourceWarning: unclosed <socket object, fd=17, family=10, type=1, proto=0>

justin1977justin, 2020-05-18:
sudo apt -y install libgpiod2

That fixed my issue. I am now able to get the temp reading, however, it
will not allow me to create a humidity sensor. It reverts back to
temperature every time.

theyosh, 2020-05-18:
Ok, made a new update.

for @justin1977justin just do a git pull for you

for @steveava remove the file terrariumGPIOSensor.py and download the new one:  wget https://raw.githubusercontent.com/theyosh/TerrariumPI/new_dht/terrariumGPIOSensor.py

and try to start.

The strange thing is that you both are saying different things based on the same code which is not possible. So make sure you all run it on Python3. Python2 will not use this library.

@justin1977justin it is unclear what you mean not able to create a new humidity sensor. Is there an error on the screen session?

I am not able to test at to moment with this sensor. So I need all information and rather precisely. So if possible provide either screenshots with the errors, or copy and paste everything...

steveava, 2020-05-18:
This is everything I get after the update.

 

pi@raspberrypi:~/TerrariumPI $ python3 terrariumPI.py

2020-05-18 21:04:36,042 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.

2020-05-18 21:04:41,087 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!

2020-05-18 21:04:46,161 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'temp'. Error message: module 'board' has no attribute 'DFalse'

2020-05-18 21:04:46,163 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'temp' is outside valid range 0.00C - 100.00C in 0.00349 seconds.

2020-05-18 21:04:56,000 - INFO    - terrariumWebserver   - Running webserver at :::8090

Traceback (most recent call last):

  File "terrariumPI.py", line 32, in <module>

    terrariumWebserver.start()

  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 560, in start

    quiet=True)

  File "/usr/local/lib/python3.7/dist-packages/bottle.py", line 761, in run

    run(self, **kwargs)

  File "/usr/local/lib/python3.7/dist-packages/bottle.py", line 3137, in run

    server.run(app)

  File "/usr/local/lib/python3.7/dist-packages/bottle_websocket/server.py", line 17, in run

    server.serve_forever()

  File "/usr/local/lib/python3.7/dist-packages/gevent/baseserver.py", line 398, in serve_forever

    self.start()

  File "/usr/local/lib/python3.7/dist-packages/gevent/baseserver.py", line 336, in start

    self.init_socket()

  File "/usr/local/lib/python3.7/dist-packages/gevent/pywsgi.py", line 1482, in init_socket

    StreamServer.init_socket(self)

  File "/usr/local/lib/python3.7/dist-packages/gevent/server.py", line 180, in init_socket

    self.socket = self.get_listener(self.address, self.backlog, self.family)

  File "/usr/local/lib/python3.7/dist-packages/gevent/server.py", line 192, in get_listener

    return _tcp_listener(address, backlog=backlog, reuse_addr=cls.reuse_addr, family=family)

  File "/usr/local/lib/python3.7/dist-packages/gevent/server.py", line 288, in _tcp_listener

    sock.bind(address)

OSError: [Errno 98] Address already in use: ('::', 8090)

2020-05-18 21:04:56,880 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'temp'. Error message: module 'board' has no attribute 'DFalse'

2020-05-18 21:04:56,881 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'temp' is outside valid range 15.00C - 45.00C in 0.00172 seconds.

sys:1: ResourceWarning: unclosed <socket object, fd=17, family=10, type=1, proto=0>

pi@raspberrypi:~/TerrariumPI $

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

justin1977justin, 2020-05-18:
Sorry, I may have spoken too soon. Previously, when I added the temperature
sensor in TerrariumPI it showed the current value as either "null" or
"none" and no sensor was added to the dashboard. After running  sudo apt -y
install libgpiod2  the sensor was added and showing a value of 20c which
lead me to believe it was working now. I attempted to add another sensor
for the humidity and after submitting, changing to dashboard, and back to
sensor settings, the sensor reverted back to temperature.
Multiple attempts at changing to humidity all reverted back to temperature.
After your email, i checked the logs and found that the error was still
present.

theyosh, 2020-05-18:
@steveava You have more then one TerrariumPI processes running. So make sure that there is nothing of my software running when we are testing... Also the value of False is strange. So what have you entered as address (in the sensor edit screen).

@justin1977justin was you able test the latest git changes?

steveava, 2020-05-18:
How do you stop the current process. Tried "sudo service terriumpi stop" from docs but it doesn't respond.

I have tried "4" "GPIO4" "GPIO 4" all give the same error. tried "17" as well just in case there was an issue with hardware.

theyosh, 2020-05-18:
Stopping using screen sessions: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi

Use pin number only as address according: https://github.com/theyosh/TerrariumPI/wiki/Hardware#gpio-pin-setup. And number 4 is a power pin, which can never be used as an sensor pin. Also number 17 is a power pin. Are you sure that you have wired it correctly.

If you did not had this running before, it will be not be wise to test this. As there are now also the problem of wrong addresses, and even broken hardware can cause problems.  So I am not sure if we can fix this all. I have to assume that had a working setup.

steveava, 2020-05-18:
Your link to stop terrariumpi is where “sudo service terriumpi stop” came from but it didn’t work.

 

pi@raspberrypi:~/TerrariumPI $ sudo service terriumpi stop

Failed to stop terriumpi.service: Unit terriumpi.service not loaded.

 

I may have figured out how to do this in screen and just to give information this is the output:

 

Unable to set line 4 to input

2020-05-18 22:11:57,736 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'temp'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-18 22:11:57,737 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'temp' is outside valid range 15.00C - 45.00C in 0.54980 seconds.

Unable to set line 4 to input

2020-05-18 22:12:27,783 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'temp'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-18 22:12:27,785 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'temp' is outside valid range 15.00C - 45.00C in 0.55652 seconds.

 

 

 

As far as pins are concerned they have been physically wired correctly but I assumed the address was GPIO number and not physical pin number. Can’t remember doing it that way before but it was a while ago. Anyway the DHT22 data pin is connected to GPIO4(pin7). I tried 7 for the address but had the same error.

 

Perhaps it would be better for me to test the GPIO just in case there’s an issue there. And perhaps do a complete re-install of Buster & TerrariumPI as I don’t understand why I couldn’t run git checkout new_dht when @justin1977justin could!

 

I’ll do these things tomorrow.

 

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

justin1977justin, 2020-05-18:
I did not see where you said to do git pull. I have done it an see no
change in regards to the error. Oddly the sensor was working outside of
TerrariumPI previously, however, after doing clean install and everything
else today it is no longer working outside of TerrariumPI. Another odd
thing is that it was showing a current value of 20c in TerrariumPI while I
had it in my office with A/C blowing on it. Now i have moved to my dining
room which is warmer and it was showing 23c and has since changed to 24.
Appears that it is getting a reading but still showing error.

I also attempted the dht_simpletest script to test it outside of
TerrariumPI and am getting the same error.

steveava, 2020-05-19:
Fresh install of Buster_lite on PI3B plus apt-get update/upgrade

sudo apt-get install python3-pip
pip3 install adafruit-circuitpython-dht
sudo apt-get install libgpiod2

Ran dht_simpletest.py for GPIO4(pin7)

This works fine and i'm getting both temperature and humidity readings from DHT22.

While I was there I tested a relay on GPIO17(pin11) and that all works fine as well.

Then:

git clone --recursive --depth 1 https://github.com/theyosh/TerrariumPI.git

All ok:

cd TerrariumPI
git checkout new_dht

And I get the same error as before: 

pi@raspberrypi:~/TerrariumPI $ git checkout new_dht
error: pathspec 'new_dht' did not match any file(s) known to git

So:

sudo ./install.sh 3 and Reboot twice.

TerreriumPI started.

Ran dht_simpletest.py for GPIO4(pin7)

and got this error:

pi@raspberrypi:~ $ python3 dht_test
Unable to set line 4 to input
Timed out waiting for PulseIn message. Make sure libgpiod is installed.
Timed out waiting for PulseIn message. Make sure libgpiod is installed.

The relay test on the other hand worked as it did before.

Added DHT22 to sensors and tried address 4 and then 7 and got the same errors as before.

After stopping TerrariumPI:

cd /home/pi/TerrariumPI
rm terrariumGPIOSensor.py
wget https://raw.githubusercontent.com/theyosh/TerrariumPI/new_dht/terrariumGPIOSensor.py
python3 terrariumPI.py

At this stage I am getting the same errors as before:

Unable to set line 4 to input
2020-05-19 10:50:51,924 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'TempTest'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

theyosh, 2020-05-19:
Thanks for the extended debug. Unfortunately, you went wrong at:

`git clone --recursive --depth 1 https://github.com/theyosh/TerrariumPI.git`

That is preventing you from changing to the new_dht branch. So this will not work.

So, I am sorry, but remove the existing TerrariumPI folder, and then do a git clone like this:

`git clone --recursive https://github.com/theyosh/TerrariumPI.git`

Without the --depth 1 part. That should give you the full source code, and when that is done, enter the folder TerrariumPI and you should be able to switch the branch. Make sure you have removed your existing TerrariumPI folder! Else the clone action will fail.

After you have switched to the new_dht branch, you have to run the installer again to install the new libraries which are needed.

But in the meantime you have found that physical pin number 7 is the correct one. So that is the value you should use as sensor address in my software. But then, how is it possible you get it working with the test code, and my version is complaining about missing libgpiod2 which you have installed as first almost.

steveava, 2020-05-19:
Ok, this time I managed to run "git checkout new_dht" and the response was that it switched. Not sure where to look to confirm that.

Anyway after a reboot and setup up the DHT22 I get the same error 

Unable to set line 4 to input
2020-05-19 16:08:33,007 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Temperature'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

And I get pretty much the same when I run "python3 dht_simpletest.py"

Unable to set line 4 to input
Timed out waiting for PulseIn message. Make sure libgpiod is installed.
Timed out waiting for PulseIn message. Make sure libgpiod is installed.

theyosh, 2020-05-19:
Ok, so now, we get the same error with my software as with the simple test of the library.

Is libgpiod2 installed? What happens when you run: `apt install libgpiod2`
Does it install software, or saying that is already installed?

justin1977justin, 2020-05-19:
I read back through and noticed that the installer was supposed to be run
after the git pull. I ran the installer this morning and am now able to see
temp and humidity readings from  "python3 dht_simpletest.py". However, now
I cannot access TerrariumPI similar to the original problem I had on the
initial build I had. I made some other changes in an attempt to get it
working after the last emails last night which may have caused problems. I
am going to do a clean install and follow the steps you provided and see if
that works now.

steveava, 2020-05-19:
Yes, so by installing TerrariumPI, that seems to stop the simple test working and gives more or less the same error in your software and the simple test.

 

Although I didn’t’ say previously, I did try to install libgpio2 thinking it was somehow removed but no, it did come back to say it was already installed. I even tried the “pip3 install adafruit-circuitpython-dht” but again it was already there and up to date.

 

 

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

theyosh, 2020-05-19:
Hmmm, this is not nice. So I have to look what my installer is doing that could cause this.

So first can you edit the file /boot/config.txt as root

`sudo nano /boot/config.txt` and look for the line: 'dtoverlay=w1-gpio' and put a '#' in front of it. This will disable the 1wire protocol. Then reboot and see if it solved the issue

justin1977justin, 2020-05-19:
I just completed doing a fresh install. Went through the steps you outlined
along the way. No change. I also tried to install libgpiod2 and was also
told that it was installed. I disabled 1wire and still no change. I get
phantom values in TerrariumPI that do not change. Still get errors in log
and same error with python3 dht_simpletest.py.

steveava, 2020-05-20:
Ok, I get slightly different results that @justin1977justin.

 

Commenting out the line in config.txt has enable the DHT22 and I am getting a Temperature and Humidity result. Interestingly the log still shows errors and it appears that your software takes an initial reading and the stops. After watching for a while I did have another Humidity value so perhaps there’s a timing issue somewhere. Humidity was high because I had my hand cupped over the sensor.

 

2020-05-20 13:47:24,133 - WARNING - terrariumSensor      - Measured humidity value 96.5% from dht22 sensor 'Humidity' is outside valid range 20.00% - 50.00% in 0.00023 seconds.

Unable to set line 4 to input

2020-05-20 13:47:54,600 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Temperature'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-20 13:47:54,602 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'Temperature' is outside valid range 15.00C - 45.00C in 0.55846 seconds.

Unable to set line 4 to input

2020-05-20 13:47:55,244 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Humidity'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-20 13:47:55,247 - WARNING - terrariumSensor      - Measured humidity value None% from dht22 sensor 'Humidity' is outside valid range 20.00% - 50.00% in 0.53795 seconds.

Unable to set line 4 to input

2020-05-20 13:48:24,632 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Temperature'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-20 13:48:24,633 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'Temperature' is outside valid range 15.00C - 45.00C in 0.55085 seconds.

Unable to set line 4 to input

2020-05-20 13:48:25,276 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Humidity'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-20 13:48:25,279 - WARNING - terrariumSensor      - Measured humidity value None% from dht22 sensor 'Humidity' is outside valid range 20.00% - 50.00% in 0.53788 seconds.

Unable to set line 4 to input

2020-05-20 13:48:54,658 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Temperature'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-20 13:48:54,660 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'Temperature' is outside valid range 15.00C - 45.00C in 0.53781 seconds.

Unable to set line 4 to input

2020-05-20 13:48:55,302 - WARNING - terrariumGPIOSensor  - Error getting new data from dht22 sensor 'Humidity'. Error message: Timed out waiting for PulseIn message. Make sure libgpiod is installed.

2020-05-20 13:48:55,305 - WARNING - terrariumSensor      - Measured humidity value None% from dht22 sensor 'Humidity' is outside valid range 20.00% - 50.00% in 0.53791 seconds.

 

I can also run the dht simple test and get results although this Checksum does seem to pop up too frequently. Perhaps timing again!

 

Temp: 83.7 F / 28.7 C    Humidity: 35.4%

Temp: 81.3 F / 27.4 C    Humidity: 37.2%

Temp: 81.3 F / 27.4 C    Humidity: 37.2%

Checksum did not validate. Try again.

Temp: 81.1 F / 27.3 C    Humidity: 37.2%

Temp: 81.1 F / 27.3 C    Humidity: 37.3%

Temp: 81.1 F / 27.3 C    Humidity: 37.3%

Temp: 81.1 F / 27.3 C    Humidity: 37.3%

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Checksum did not validate. Try again.

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Checksum did not validate. Try again.

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Temp: 81.1 F / 27.3 C    Humidity: 37.4%

Checksum did not validate. Try again.

Temp: 81.0 F / 27.2 C    Humidity: 37.4%

 

 

Just to be sure I removed the # and it stopped working again both in your software and simple test.

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

steveava, 2020-05-20:
Update:

I'm getting changes in Temperature and Humidity values occasionally.

theyosh, 2020-05-20:
Hmm, strange. I am currently trying to use another different library. But it takes some time to fix my setup. So thanks for the update, and hang on, I hope to have an update soon (today)

theyosh, 2020-05-20:
Ok, so it should now be fixed. I have changed the code, and removed the Adafruit libraries. Now I use a different type of approach, which is probably also fix some Raspberry PI Zero issues when reading GPIO pins.

So so `git pull` and restart. That is all. Your DHT sensors should work again! At least I got some data, but it seems that the humidity is broken at my sensor... I hope.. So give it a try

steveava, 2020-05-20:
That seems to have done it.
Had to remove and recreate the sensors but all seems to work now and no errors.

Thanks.

theyosh, 2020-05-20:
I hope that @justin1977justin also can confirm that it is working form him. That would mean that both DHT22 and DHT11 are working again.

justin1977justin, 2020-05-20:
I will check later tonight. I am finishing the wiring of my case and took
the dht11 offline last night. I will get it finished tonight and let you
know.

justin1977justin, 2020-05-20:
python3 dht_simpletest.py works and shows readings that update properly. I
am no longer getting the error that libgpiod is not installed in
TerrariumPI. I am still unable to get sensor values in TerrariumPI.

[image: image.png]

justin1977justin, 2020-05-21:
After 15-20 mins i noticed that there were values showing for the Temp
sensor but not the Humidity sensor. At first the temp was pretty similar to
my other temp sensors. Then the temp changed to 32. I'm attaching a
screenshot showing the values from dht_simpletest.py vs what is shown in
TerrariumPI.

[image: image.png]

theyosh, 2020-05-21:
Images attached to emails are not showing here....

theyosh, 2020-05-21:
Ok, after more searching, I found out that manual installing the old/existing Adafruit DHT driver, it will work. https://github.com/adafruit/Adafruit_Python_DHT/issues/44

So, I reverted back all the changed DHT code, and changed the installer. So could you test this, and see if this will work for you.

`git pull`
`sudo ./install.sh 3`
`python3 terrariumPI.py`

I still assume you have a DHT sensor configured in your settings file.

justin1977justin, 2020-05-21:
I am running installer now. Odd thing is that the sensors showed 32 for temp and 0 for humidity all night until around 7am when it began showing the correct readings. I did not attempt to see if it would register changes correctly. I will update after installer is finished.

steveava, 2020-05-21:
Not sure this has helped for my install.

 

Back to this error again when running dht simple test.

 

Unable to set line 4 to input

Timed out waiting for PulseIn message. Make sure libgpiod is installed.

Timed out waiting for PulseIn message. Make sure libgpiod is installed.

 

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

steveava, 2020-05-21:
But getting results in TerraruimPI! Temperature seems to be intermittent though.

2020-05-21 13:48:13,139 - INFO    - terrariumSensor      - Updated bme280 sensor 'Hum2' humidity from 29.63% to 29.63% in 0.00018 seconds
2020-05-21 13:48:13,018 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temp2' temperature from 29.00C to 29.00C in 0.00020 seconds
2020-05-21 13:48:12,895 - INFO    - terrariumSensor      - Updated dht22 sensor 'Hum' humidity from 36.40% to 36.40% in 0.00018 seconds
2020-05-21 13:48:12,778 - INFO    - terrariumSensor      - Updated dht22 sensor 'Temp' temperature from 28.60C to 28.50C in 0.00033 seconds
2020-05-21 13:48:12,353 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
2020-05-21 13:47:58,271 - INFO    - terrariumEnvironment - Update done in 0.00231 seconds. Waiting for 14.99769 seconds for next update
2020-05-21 13:47:54,934 - INFO    - terrariumEngine      - Update done in 12.22848 seconds. Waiting for 17.77152 seconds for next update
2020-05-21 13:47:54,790 - INFO    - terrariumSensor      - Updated bme280 sensor 'Hum2' humidity from 29.97% to 29.63% in 0.00033 seconds
2020-05-21 13:47:54,671 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temp2' temperature from 29.01C to 29.00C in 1.01527 seconds
2020-05-21 13:47:53,536 - INFO    - terrariumSensor      - Updated dht22 sensor 'Hum' humidity from 36.90% to 36.40% in 0.52801 seconds
2020-05-21 13:47:52,850 - WARNING - terrariumSensor      - Measured temperature value NoneC from dht22 sensor 'Temp' is outside valid range 0.00C - 100.00C in 10.11469 seconds.
2020-05-21 13:47:43,266 - INFO    - terrariumEnvironment - Update done in 0.00214 seconds. Waiting for 14.99786 seconds for next update
2020-05-21 13:47:43,262 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2020-05-21 13:47:42,337 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
2020-05-21 13:47:28,258 - INFO    - terrariumEnvironment - Update done in 0.00218 seconds. Waiting for 14.99782 seconds for next update
2020-05-21 13:47:13,241 - INFO    - terrariumEnvironment - Update done in 0.00113 seconds. Waiting for 14.99887 seconds for next update
2020-05-21 13:47:13,235 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
2020-05-21 13:47:13,185 - INFO    - terrariumEngine      - Update done in 0.52025 seconds. Waiting for 29.47975 seconds for next update
2020-05-21 13:47:13,056 - INFO    - terrariumSensor      - Updated bme280 sensor 'Hum2' humidity from 29.97% to 29.97% in 0.00017 seconds
2020-05-21 13:47:12,932 - INFO    - terrariumSensor      - Updated bme280 sensor 'Temp2' temperature from 29.01C to 29.01C in 0.00020 seconds
2020-05-21 13:47:12,810 - INFO    - terrariumSensor      - Updated dht22 sensor 'Hum' humidity from 36.90% to 36.90% in 0.00018 seconds
2020-05-21 13:47:12,694 - INFO    - terrariumSensor      - Updated dht22 sensor 'Temp' temperature from 28.60C to 28.60C in 0.00019 seconds

justin1977justin, 2020-05-21:
Last git pull has fixed the issue. I am now getting the same readings from dht_simpletest and TerrariumPI. Also no errors or warnings in logs.

justin1977justin, 2020-05-21:
@theyosh Thanks for working through this with us to get it straightened out.

theyosh, 2020-05-21:
@steveava DHT sensors are not that stable. So getting an error once a while is not an issue.

And thanks for testing. After all it was just an update of the installer :( So we tried a lot of code, but got back by the old libraries. Lets hope they stay working now...

I have merged this to the master branch. So if you now switch back to the master branch then you will stay up to date:

`git checkout master`
`git pull`

steveava, 2020-05-21:
Ok, I'll switch back.

Thanks.

theyosh, 2020-05-22:
So, 24 hours later, it is still working?

justin1977justin, 2020-05-22:
Still working for me.

steveava, 2020-05-23:
Yep, still working here.

theyosh, 2020-05-23:
Closing now. Thanks for testing!

closed 2020-05-23 by theyosh