theyosh/TerrariumPI #393: Question about dimmers

martynwheeler, 2020-05-14:
Good morning,

Not an issue, just a bit of clarification.  In the setting for my MPDMv3 MAINS Dimmer there are settings for "dimmer on duration" and "dimmer on percentage".  Do these control what happens when the heat is switched on at the beginning of the day?  I am finding that the it is going to 100% quite quickly and then overshooting the desired temperature.  I realise that this is because it is having to warm quite a big volume.  Could you please explain the function of these settings as I am not clear?  Thanks.

theyosh, 2020-06-01:
Hi @martynwheeler 

it took a bit more time, due to other more demanding issues.

First the options 'dimmer on/off percentage' are values that specify the min and max dim values. So you could say that a dimmer is off when the dimm value is at 10%. And when using a dimmer as heater for example, you can set a max dim value. So the dimmer will never go beyond that value.

The duration is the time the dimmer should take when we put it on. So from off to on will happen in this amount of seconds.

This was intended to be used when you out the dimmer just on or off like normal switches.

But, at this time, I use a different logic for the heating. It uses the value of  'Dimmer environment step' with 'Dimmer action duration'. Every time the heater should increase based on sensors, it will increase with this value in the specified duration.

In a newer function, this logic will change again I think. As this currently a bit confusing, and hard to calculate the sun set or sun rising logic with the dimmer.

martynwheeler, 2020-06-01:
Thanks for the explanation, I will close the issue.

closed 2020-06-03 by theyosh