theyosh/TerrariumPI #394: Restarting in 10 seconds after running for 18 seconds. :/

blackyarr, 2020-05-14:
Clean instalation on raspberry pi zero W

ImportError: No module named luma.core.interface.serial
2020-05-14 14:31:07,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2020-05-14 14:31:08,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 18 seconds.

after 5 screen is terminatig

Any ideas how to fix it?

theyosh, 2020-05-16:
Hmm, sounds like you are missing a library.

And the Pi Zero is a bit strange. So you have to run everything as user root. Also the installation and git checkout needs to be done on the /root location.  This way, everything will run as root which is needed.

So if you do a another install in the /root folder and reboot. Let see if it works. If the library is still missing, we have do to some manual actions.

But, if I may give a bit of advise, switch to a Pi4. That is more stable and easier to work with. Also the installation does go faster.

theyosh, 2020-05-21:
Ok, found out that this is happening when you install the software with Python2. Please use Python3 to install and run this software.  https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

blackyarr, 2020-05-21:
I apologize for the silence in the subject.
Thank you for your answer.
I also came up with a solution using phyton 3 unfortunately I didn't have time to test it.
I would prefer not to run the application in the root folder.

theyosh, 2020-05-21:
With a Raspberry Pi Zero, the gpio rights are different, and you get a lot of errors. Therefore the software needs to run as root. And this is triggered by installing the software in a folder with root rights.

But the point stays, it needs to run as root, which I do not like. Also, it is pretty slow compared to a Pi4. So if the root issue is a concern, I advise to go for a Pi4 with Python3. That works the best at the moment.

blackyarr, 2020-05-21:
Ok in Python3 i got success

----
2020-05-21 16:22:42,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.8.195:8090 ...
2020-05-21 16:23:46,424 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2020-05-21 16:23:51,583 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2020-05-21 16:24:03,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
---

Thank you for the tips. I suppose it will move to Pi4 but I would like to test how it works and Pi0w was on the shelf.

theyosh, 2020-05-22:
So we close this issue?

theyosh, 2020-05-23:
Closing due to working now. If there are still issues, you can re-open this one of start a new one.

closed 2020-05-23 by theyosh