theyosh/TerrariumPI #396: 1wire sensors stopped working

justin1977justin, 2020-05-24:
I have 2 separate builds running. 1st has been running for several weeks without issue. 2nd was just recently setup but temp sensors were working great. Noticed sensors stopped working on 1st build yesterday. I restarted pi and could no longer access through UI or ssh. I power cycled the pi after a few mins and the system was restored. Looking at the sensor graph on system 1, i noticed that the sensor stopped working again over night for several hours then started working again. Now around 5pm local time all sensors on both systems stopped working. Receiving same error message on both except for s/n of sensors. Originally though sensor was going bad, however, i think it is unlikely that 4 sensors all went bad at the same time.

2020-05-23 22:08:54,761 - WARNING - terrariumSensor      - Measured temperature value NoneF from w1 sensor 'Cool Hide' is outside valid range 55.00F - 95.00F in 0.00173 seconds.
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-3c01b556a2ea/w1_slave'
with open(os.path.join(terrarium1WSensor.W1_BASE_PATH,self.get_address(),'w1_slave'), 'r') as w1data:
File "/home/pi/TerrariumPI/terrariumSensor.py", line 330, in load_data
Traceback (most recent call last):
2020-05-23 22:08:54,760 - ERROR   - terrariumSensor      - Error loading 1 Wire data at location: /sys/bus/w1/devices/28-3c01b556a2ea/w1_slave with error: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-3c01b556a2ea/w1_slave'
2020-05-23 22:08:54,651 - WARNING - terrariumSensor      - Measured temperature value NoneF from w1 sensor 'Warm Hide' is outside valid range 65.00F - 100.00F in 0.00260 seconds.
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-3c01b556029c/w1_slave'
with open(os.path.join(terrarium1WSensor.W1_BASE_PATH,self.get_address(),'w1_slave'), 'r') as w1data:
File "/home/pi/TerrariumPI/terrariumSensor.py", line 330, in load_data
Traceback (most recent call last):
2020-05-23 22:08:54,649 - ERROR   - terrariumSensor      - Error loading 1 Wire data at location: /sys/bus/w1/devices/28-3c01b556029c/w1_slave with error: [Errno 2] No such file or directory: '/sys/bus/w1/devices/28-3c01b556029c/w1_slave'

justin1977justin, 2020-05-24:
Ok, I have Home Assistant set to alert me through Amazon Echos when temp is low on the first build. Randomly, it alerted me so i checked and sure enough both sensors were working again. Nothing had been done on that pi for at least 30 mins. Last thing done was a power cycle. I ran the installer again on build 2 hoping that it would repair the error. It did not, still had the same error. Power cycled that one and it started working instantly. Any idea what may be causing the issue? Reboot does not correct it. Only a power cycle appears to.

martynwheeler, 2020-05-24:
Sounds similar to issues described here

https://www.raspberrypi.org/forums/viewtopic.php?t=164059

The fix is to power the 1 wire sensor from a gpio output pin rather than
the 3.3V pin. You can the simply cycle the power on the one wire bus by
switching the state of the pin. There is a script on the page above to
automatically detect if the sensor has stopped responding and to attempt a
reset.

Martyn

theyosh, 2020-05-24:
Hmm, if 1wire is failing that means that it is failing on the OS level. I cannot fix that from my software side. But I have experienced also issues with unstable sensors when the cables are long, or when I put a lot on 1 I2C bus. Mostly it is an power issue.

And I do know that the power of Pi is limited, and for long wire systems, use an external power source, which are more stable. But I do not have experience with.

So, based on the above forum thread, I can make some 1Wire reset code. But that will be something that has to setup manually. As this means that the power of the 1Wire is not coming from the normal 3.3V power pin, but from a random GPIO pin. So that is something that needs to be setup by the user.

theyosh, 2020-05-24:
Do you have by any change low power messages when run `dmesg` ?

theyosh, 2020-05-24:
Hmm, maybe it is easier to use: https://github.com/SkullKill/onewireMonitor

theyosh, 2020-05-24:
And some remarks from the above repository:

> rpi has a limit of 16mA per gpio, with a total of 51mA for all gpio.
>
> the DS18B20 sink a max of 4mA , so one gpio can technically power a maximum of 4 , although to be safe, might be better to stick to a max of 3

justin1977justin, 2020-05-24:
Build 1 is getting under voltage messages. I am working to make my final box for build one and will be incorporating a proper power supply. I will see if that resolves it. Another thing i read was that power fluctuations on the same circuit can cause this. This did not start until I added the second build which is connected to the same outlet. I may need to isolate the 2. If power supply or isolating doesn't work I will try the other method listed.

martynwheeler, 2020-05-24:
You can also power 1 wire from 5V if using a long wire, you will just need
to ensure that you have a logic level shifter to bring the signal down to
3.3V before putting back into the Raspberry Pi Gpio.

theyosh, 2020-05-24:
@martynwheeler Do you have any experience with that....? As I also red about using resistors as a simple voltage divider https://en.wikipedia.org/wiki/Voltage_divider 
But I do not have experience with this.

martynwheeler, 2020-05-24:
I have not done it with 1-wire but have used level shifters in other
applications. You could use voltage divider but since level shifters are so
cheap I would do it that way (eg
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.sparkfun.com/products/12009&ved=2ahUKEwiDkd-klc3pAhXJMMAKHUf7ATAQFjABegQICRAB&usg=AOvVaw09l-95tsDFiTyHqnaFBelE
)

Just make sure your 4.7k pull up resistor is on the 5V side of the level
shifter.  I can give it a test on a spare pi to check the wiring.

Martyn

martynwheeler, 2020-05-24:
Actually having thought a little more. The simplest solution to power your
ds18b20 using 5V is to do the following:

Connect Vdd of sensor to 5V and gnd to gnd on RPi. Then connect 1-wire
signal line (pin 7) to 3.3V using a 4.7k pull up resistor. This way you can
use longer cable lengths and not suffer voltage drops that might cause the
sensor to be lost.

This way you don't need any extra hardware.

Hope this helps.

Martyn

justin1977justin, 2020-05-25:
@martynwheeler Thanks. I may actually already have it connected that way. It will not work at all on my 1st build now. Even after power cycle. 2nd build has been working fine since I ran the installer again yesterday. I am currently running updating this one as well just to eliminate that as a possible cause. Could it possibly be a hardware failure with the PI? I have it running as a temporary setup right now to figure out how i wanted my box. I am working on finishing the box for this one and may just get a new pi and do a whole new build to eliminate the possibility of the pi being the issue. The pi is older and is running hotter than i would like so wouldn't be the end of the world to replace it. I also will test additional temp sensors to see if they are possibly the cause.

martynwheeler, 2020-05-25:
If you are running at 5V and the signal is at 5V you can damage the gpio
pin. Might be worth checking the pin is okay still.

justin1977justin, 2020-05-25:
Well, this morning I could not access pi anymore after running installer again. Pulled pi from box and updated wpa_supplicant and ssh and was able to access again. Still no temp sensors registering. changed to another pin and update config and so far so good. I am going to close issue for now.

closed 2020-05-25 by justin1977justin