theyosh/TerrariumPI #397: New Install RPI zero w - new buster 2020-20 python3 website doesn't start

se7en9057, 2020-05-25:
So i understand that there is an issue with the RPI0w and ive worked through to get it working however i have to get it to start manually. i'm running the newest buster lite with python 3 following the instructions in the KB's. 

have to start the service with sudo and do a systemctl daemon-restart among other things. can anyone share any detailed instructions on how to install TerrariumPi on a Zero from scratch. i'm not a noob with it comes to linux but i'm rusty. I know this sort of issue has been gone over in bits and pieces but it would be convent if there was a full install guide for a pi zero. i'm happy to help with that if we can collaborate. just trying to get this thing to run from boot up and auto start the service with out having to ssh in and tinker with it to get it to work. 

I can get my DHT11 sensors to read data and i'm waiting on my relay board to get in to test with that as well. 

thanks in advance for any redundancy this may be.

se7en9057, 2020-05-25:
done this so far however during boot the console says that terrariumpi is started as user "pi" still not able to access the site http://<myip>:8090

1. install fresh copy of rasbian buster 2020-2
2. sudo apt update
3. sudo apt install git
4. sudo -i
4b. passwd 
	(create root user password)
5. cd /root
6. git clone --recursive --depth 1 https://github.com/theyosh/TerrariumPI.git
7. cd TerrariumPI
8. ./install.sh 3 
(you should still be sudo so no need to sudo before this command)
8b. syste reboot
8c. ssh back into pi 
8d. sudo -i 
9. cd /root/TerrariumPI
10. ./start.sh 3

	And change the command in /etc/rc.local from
	/home/pi/TerrariumPI/start.sh
	to
	/root/TerrariumPI/start.sh 3

theyosh, 2020-05-26:
Hi,

after step 10, you could just do a restart. Just to make sure the Python3 version is running.

After the restart, become root again, and then try `screen -r` It should give you a debug screen.

If not, do a `ps fax` and see if there is a python3 process running terrariumPI.py

if that is also not the case, try running it manually:

`cd /root/TerrariumPI`
`python3 terraiumPI.py`

I guess the last command will give an error... and that is what I need. Else I can't tell what is going on.

se7en9057, 2020-05-26:
Doesn't look like terrariumpi.py is running checked as pi and as sudo/root by ps fax

running manually
sudo -i
cd /root/TerrariumPI
python3 terraiumPI.py
python3: can't open file 'terraiumPI.py': [Errno 2] No such file or directory

well i guess it can't run if its not there. interesting...
i'll include the ps fax dump for reference. 
thanks. 
[psfaxterrariumpitroubleshooting_1.txt](https://github.com/theyosh/TerrariumPI/files/4685131/psfaxterrariumpitroubleshooting_1.txt)

theyosh, 2020-05-26:
I am sorry, but this is not clear. Are there files in the TerrariumPI folder? Did the git checkout give some messages? Because you have the file ' terrariumPI.py'

se7en9057, 2020-05-26:
sorry spelling error. 

root@raspberrypi:~/TerrariumPI# python3 terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/root/TerrariumPI/terrariumEngine.py", line 32, in <module>
    from terrariumSwitch import terrariumPowerSwitch
  File "/root/TerrariumPI/terrariumSwitch.py", line 1211, in <module>
    class terrariumPowerSwitch(object):
  File "/root/TerrariumPI/terrariumSwitch.py", line 1240, in terrariumPowerSwitch
    from terrariumSwitchKasa import terrariumPowerSwitchTPLinkKasa
  File "/root/TerrariumPI/terrariumSwitchKasa.py", line 11, in <module>
    from kasa import Discover, SmartStrip, SmartPlug
ModuleNotFoundError: No module named 'kasa'
root@raspberrypi:~/TerrariumPI#

se7en9057, 2020-05-26:
also if its easier for troubleshooting i can set up a screen share.

se7en9057, 2020-05-27:
root@raspberrypi:~/TerrariumPI# ./start.sh
2020-05-26 20:26:01,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server running as user 'pi' at location: http://192.168.2.104:8090 ...

theyosh, 2020-05-27:
Hmm, you are mixing python2 and python3.

you should run: `./start.sh 3` because you have installed with Python3.

And then, you can run `screen -r` to see what the software is doing.

se7en9057, 2020-05-27:
i guess that was my fault on the manual command. attached is the psfax output screen -r says there is no screen to be resumed. 
[psfaxterrariumpitroubleshooting_2.txt](https://github.com/theyosh/TerrariumPI/files/4688379/psfaxterrariumpitroubleshooting_2.txt)

theyosh, 2020-05-27:
There is a screen session running according to your txt file. So there should be a screen to be resumed.

Did you try it as root? Maybe try it as user pi. Again, installing on a PiZero give all kind of issues, and mostly my answer is: Use a Pi3 or Pi4. They are proven to work well.

se7en9057, 2020-05-27:
roger that. strange the pi zero gives so many issues. i had it working briefly but had to manually start it among other things. would be cool to have an official pi zero support channel. i may just break down and pull my pi 3 from the other project and replace it or just buy another one :P kinda feel like thats not really fixing the issue just buying a solution. but i also agree its your project and your donating your time. is there any way to help get the pi zero support? anything i can do?

se7en9057, 2020-05-28:
I ended up grabbing a pi 3b+ so i'll shelve the pi zero for now. i'm happy to help with getting a pi zero running on this platform tho. just let me know. i'm having some issues with the 3b+ but i'll do some more digging before i open another request or issue.

closed 2020-05-28 by se7en9057