theyosh/TerrariumPI #398: Kasa Smart Wi-Fi Plug Mini

jaysalazar, 2020-05-25:
I noticed you added support for TP-link Kasa Smart Strip. Will the Kasa single plug mini (HS-105) work as well or will this require additional work for integration?

theyosh, 2020-05-26:
I had to add some small part of code. If you update now, it should work.

Do a `git pull` and restart. It should find your device automatically. If not, enter the IP address of the switch when manual adding a new switch

jaysalazar, 2020-05-26:
Thanks for the quick respsonse. I backed up my settings and did a git pull however I am getting the following after the update/restart; 

2020-05-26 14:53:32,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 5 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2020-05-26 14:53:43,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.1.60:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 32, in <module>
    from terrariumSwitch import terrariumPowerSwitch
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 1211, in <module>
    class terrariumPowerSwitch(object):
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 1240, in terrariumPowerSwitch
    from terrariumSwitchKasa import terrariumPowerSwitchTPLinkKasa
  File "/home/pi/TerrariumPI/terrariumSwitchKasa.py", line 11, in <module>
    from kasa import Discover, SmartStrip, SmartPlug
ModuleNotFoundError: No module named 'kasa'

jaysalazar, 2020-05-27:
Update: I just installed on a new RP4/SDCARD and getting the same error message. I am running it on Raspbian Buster with desktop and recommended software, which it what my other Pi had been running for a couple months.

theyosh, 2020-05-27:
Hmm, turns out that the installer for the Kasa library has changed.

So, if you do now a `git pull` and rerun the installer, it should install the Kasa library. It will take some time, as the installer is building the module on the fly.

After the installer has finished, you can test if the Kasa is installed correctly:

`kasa --version`

should give something like: `kasa, version 0.4.0.dev0`

jaysalazar, 2020-05-27:
It is working on my test environmnet after the git pull (also auto found my tplink smart plug!), however my production environment has no change. I am now re-running the installer now.

jaysalazar, 2020-05-27:
Update: I found that I was missing the "dist" folder in the "python-kasa" directory. I manually placed it there and ran the install again and it is up now.

theyosh, 2020-05-27:
The 'dist' folder should be created by the installer. So did you do a `git pull` and then `./install.sh 3` again? Because the installer is updated in order to install the library.

But if it is working now, I hope it stays working :)

jaysalazar, 2020-05-27:
Yes, I did git pull and then ./install.sh 3 like three times total. Yes it working now, I also noticed how much faster the Pi 4 installs/runs so I may replace my hardware soon anyways :) 

Thank you very much for the added TPLink SmartPlug support and help with the kasa installer.

theyosh, 2020-06-01:
Still working? Can we close this issue?

jaysalazar, 2020-06-01:
Yes, looks like everything is still working, thank you very much. I swill now close.

closed 2020-06-01 by jaysalazar