theyosh/TerrariumPI #399: No sound card in settings drop down - unable to modify settings

se7en9057, 2020-05-29:
No sound card shows up in the drop down in settings causing me to not be able to modify settings. 
Fresh install
installed as sudo ./install.sh 3
reboot
cleared browser cache
reboot
added ip to java exceptions in java config
added ip to exceptions in browser
still no settings

theyosh, 2020-05-29:
Is this on a Pi3 or Pi4? I just saw this also today. I have also issues with mu I2C. And there is a new kernel and some changes in hardware... :(

Also, there are some alsamixer issues... which could be case of this. So I will have to look into this.

theyosh, 2020-05-29:
It is an issue with my software. So not your browser or java settings....

se7en9057, 2020-05-29:
brand new Pi 3b+ on fresh install. 
sensors seem to work relays seem to work so far DHT11's just for update off my other issue with the zero. 
thanks for the quick responce. i'll keep tinkering with the GPIO and accessories till i get an update from you.

justin1977justin, 2020-05-30:
I'm having the same issue. I was getting a 500 error and having i2c issues so I did a clean install and now i can't update settings saved because there is no sound card. At least it's not just me.

theyosh, 2020-05-30:
Do a `git pull` and restart. It should work again.

luke5sky, 2020-05-30:
worked for me, thank you!

se7en9057, 2020-05-30:
awesome. seeing sound cards now!

closed 2020-05-30 by se7en9057