theyosh/TerrariumPI #400: After initial setup unable to access web page

se7en9057, 2020-05-30:
Did base install following detailed instructions. 
added 2 DHT11 sensors
change from metric to imperial 
added location
saved. sensors seem to read base line temp
restart Pi
no unable to access web page.

se7en9057, 2020-05-31:
adding screen -r dump
[issue400_1.txt](https://github.com/theyosh/TerrariumPI/files/4706923/issue400_1.txt)

se7en9057, 2020-05-31:
might be something in the terrrariumpi.py audio engine not sure. i had mine set to the head phone jack.

codydeluisio, 2020-05-31:
hmm I had a fresh install yesterday, then was hit with the unable to change settings, git pulled earlier today, changed settings, now just got home and experiencing this too. 

Restarts aren't fixing it. I'm going to keep trying things and report back if I find anything that helps.

codydeluisio, 2020-05-31:
Well, I was able to delete the headphones line from settings.cfg and replace it with bcm2835 ALSA and now have my greenhouse back up. 

This is on a pi 4

Here are my screen logs from before though in case it helps: 

2020-05-30 22:06:10,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.165:8090 ...
2020-05-30 22:06:14,702 - WARNING - terrariumEngine      - Unable to get the latest version information from Github. Will check next round.
2020-05-30 22:06:15,189 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2020-05-30 22:06:18,616 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 27, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 164, in __init__
    self.get_audio_playing)
  File "/home/pi/TerrariumPI/terrariumAudio.py", line 45, in __init__
    self.__load_audio_mixer()
  File "/home/pi/TerrariumPI/terrariumAudio.py", line 78, in __load_audio_mixer
    self.__audio_mixer = alsaaudio.Mixer(control='PCM',cardindex=self.__hwid)
alsaaudio.ALSAAudioError: Unable to find mixer control PCM,0 [hw:0]

se7en9057, 2020-05-31:
@codydeluisio did you do that in a .conf or in the terrariumaudio.py its self?

se7en9057, 2020-05-31:
yeah its for sure something in that audio device setting. personally i'll probably never use it so now that all my units are situated i'll probably just leave it alone. still probably a bug related to the last update @theyosh could be wrong but seems logical.

codydeluisio, 2020-05-31:
I did this in the .conf. Now that I have the settings setup I'll never change them too.

theyosh, 2020-06-01:
Sorry guys... apparently the alsa mixer has changed. It should now be fixed that at least everything will start normally. I did not test if it still plays audio. I am not able to test that at the moment.

So, now you can select the Headphones soundcard, and that will not break the startup / reboots anymore.

Do a `git pull` and restart

se7en9057, 2020-06-02:
working so far so good :)

closed 2020-06-02 by se7en9057