theyosh/TerrariumPI #401: Notifications idea / request

ruepax, 2020-05-31:
Hello theyosh,

Thanks so much for this amazing code, i was thinking of adding the following feature to the notifications tab. In my opinion it would make sense to create a timing rule for the sensors alarms.

At present if temp/humidity etc is above the range you will get a notifications ever few seconds, is it possible to set up a timer on the notifications from the sensors? Like the program will only scan for notifications every 15min, and every 15m you get the last status from the sensors if is above a range.

Ideally we could set up different times for sync on the notification page.

I will try and give it a go next week, but it will be my first "python project" and I've recently started learning.

theyosh, 2020-06-03:
Hi,

this sounds like a bit like: https://en.wikipedia.org/wiki/Email_digest

This will involve settings per notification if you want to have it directly or summarized... Because I can imagine that errors will not be summarized but send directly. Warnings could be sent every 15 min or so.

I think that you still need to send all the warnings and error messages. So you should record the messages for the specified time and then sent them at the next message that is outside the time value.

I think there will be some more logic to at to make it work. So I am not sure how easy this will be, or how flexible we can get.

theyosh, 2020-06-12:
Hi,

if you are just learning Python, then my software is not the right starting point. As I use a lot of parallel processing either with  thread or geven events, it is a bit hard to understand for a starting to learn python.

And therefore, using sleep functions is not the right way. That will block also other requests, and other processes. So that will not work.

So, maybe I will add such a thing later. As the current notifications is also not optimal, due to the fast/simple implementation. As soon as you give options to add digest options to group messages, everybody wants something different. So that will introduce more options, and more logic.

So for now, I will not add it. As the software is getting pretty stable, and would like to leave it that way. I am thinking of a new rewrite of the code, so that I can add some waiting wishes..

theyosh, 2022-01-27:
Hi, it took some more time, but I have thought about this, but I have to say, I cannot implement this. It will add to much extra logic and code which goes beyond the idea of this software. Automating a terrarium.

So, sorry to say, but I will not implement this feature.

closed 2022-01-27 by theyosh