theyosh/TerrariumPI #402: 500 internal server error  - IOError(24, 'Too many open files')

nastea1, 2020-06-03:
Error: 500 Internal Server Error
Sorry, the requested URL 'http://192.168.0.16:8090/' caused an error:

Internal Server Error
Exception:
IOError(24, 'Too many open files')
Traceback:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 868, in _handle
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1748, in wrapper
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 44, in webserver_headers
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 86, in wrapper
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 240, in __render_page
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 229, in __template_variables
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1326, in get_profile_name
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1317, in get_profile_config
IOError: [Errno 24] Too many open files: 'description.txt'

theyosh, 2020-06-03:
run as normal user `lsof -u pi` and it should give a list of open files or connections.

If the list of /dev/i2c is high or something with gpio. Look at the screen session to see what the errors are. I guess a faulty sensor.

nastea1, 2020-06-03:
@theyosh Thanks it was i2c, a sensor I never got around to setting up fully hahahaha. Thanks.

closed 2020-06-03 by nastea1