theyosh/TerrariumPI #403: How to read out analog capacitive soil moisture sensor in terrariumPI

timmakken, 2020-06-05:
Hi yosh, 

let me begin with saying thanks for all your hard work on this project! its really awesome!
Ive been using the ADCmodule ADS1115 with a capacitive soil moisture sensor and have gotten some data back using a tiny python script.

result:
`--------------------
Channel 0 =2.801000V raw=0x576D dec=22381
Channel 1 =0.614125V raw=0x140A dec=5130
Channel 2 =0.551375V raw=0x12D9 dec=4825
Channel 3 =0.686250V raw=0x1205 dec=4613
--------------------`

the ADS1115 is attached to the PI on pin 3 /GPIO2 but im currently stuck on how to add this sensor to terrarium pi and get back some usefull information.

do you have any tips on how to proceed from here?

theyosh, 2020-06-12:
Hi, thanks!

I think this is possible in 2 ways. But first, you need to make your current python script a bit different. You have to change your current script in such a way, that it will only print the current moisture value in percentage. So it should be a number from 0 till 100.

If the script can be run with the normal pi user, you can then add a sensor with the hardware type custom script'. As address you enter the full path to the script. And select moisture as type.

Or, and this is a bit harder to do, you have to write the moisture value into a json file. And that file needs to be placed in the folder '/home/pi/TerrariumPI/static/extern' so that you can use it as a 'remote sensor' and then use the address like: 'http://127.0.0.1/extern/[your.json_file]'. There is an example json file in the extern folder.

So, I think, you have to update your script so that it will just print the current moisture value based on the voltage values, and then use 'custom script' sensor type.

To get the right formula from voltage to moisture, I do not know. I think you should Google for it

theyosh, 2020-06-24:
Hi, any updates? Is this working for you? Need some more help?

theyosh, 2020-07-03:
Any updates? Can we close this issue?

timmakken, 2020-07-09:
Hi Yosh! thank you for the extensive reply. i havent had a chance to try it out yet but this issue can be closed. i will soon be trying out the distance sensor to measure water level. If i get it all up and running i can add some documentation on the matter to make it easier for future users to get started with these sensors ive been using. 

Kind regards!

closed 2020-07-09 by timmakken