theyosh/TerrariumPI #404: Setting up an oled display

Dave0408, 2020-06-09:
Hello, 

i have a 128 x 64 pixel 0.96 inch OLED I2C display with SSD1306 chip.
I try to integrate it into TerrariumPi, but I can't find the right place for configuration. 

```
pi@raspberrypi:~ $ i2cdetect -y 1
     0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
```

A short test of the screen with an external library was successful. 

I have tried in /TerrariumPI/settings.cfg 

```
[display]
address = 3c,1
title = true
hardwaretype = SH1106
```
(https://github.com/theyosh/TerrariumPI/issues/232#issuecomment-453862721)

and restart the server. But he doesn't recognize it. 
I have seen a screenshot showing a configuration panel via the website.(https://github.com/theyosh/TerrariumPI/issues/193#issuecomment-455760613)
Unfortunately I cannot find it in my version.

I have the following setup:
Raspberry Pi Model B Plus Rev 1.2  
TerrariumPI 3.9.8
4x DS1820
1x BME280
1x 4 relay modules
1x OLED display

Thanks in advance for any help.
I'm pretty sure I'm just too blind to see it. ;)

theyosh, 2020-06-12:
Hm, you should be able to setup a display on the Notifications page.

There you can select your version and address. And 3c is correct as an address.

The display is only used for showing notification messages. And is not that helpfull to be honest.

Most people make there own small program that uses my API to readout information and make there own display code.

Dave0408, 2020-06-16:
Thank you very much. As I suspected, I was simply too blind. 
I'll just close this issue real quick.

closed 2020-06-16 by Dave0408