theyosh/TerrariumPI #406: How to control heat mat

darrynl, 2020-06-13:
Good Day All,

First first of all, I would like to thank @theyosh for a great piece of work.  Thank you very much for doing this.

I am new to TerrariumPi and I need some assistance.
I would like TerrariumPi to control a heat mat, based on the temp readings of a sensor.
I have managed to get the sensor working.
I have managed to get the relay to switch on and off in the switches menu.
My queston is, how to I set things up so that the heat mat is switched on and off based on the temperature reading of the sensor.  ie. I want to set it up like a thermostat.
If you could point me in the right directorion, I would be very grateful.

Regards
Darryn

se7en9057, 2020-06-14:
@theyosh i think we need a discord for this project. 
@darrynl i'm also new but ive been tinker for a few weeks now and have lots of things running. hopefully i can help.

- go to system>environment>temperature 
- ensure you mode is "sensor" and select your sensor
- if there is a day/night difference set it and source
- then set the high and low alarm modes. those values are configured in the sensor [settings.](https://i.imgur.com/Tf1qVgt.png)
- [https://i.imgur.com/pHuEJHc.png](https://i.imgur.com/pHuEJHc.png)

darrynl, 2020-06-21:
Thank you @se7en9057 
I appreciate your input.
I seemed to have got it working.

closed 2020-06-21 by darrynl