theyosh/TerrariumPI #407: Restore Backup Problem

kaju666, 2020-06-17:
Hi

My SD broke.
After install on Buster with python 3 i have problem with restoring settings.cfg
When i copy it to the folder TerrariumPi won't start giving me this error
[settings.txt](https://github.com/theyosh/TerrariumPI/files/4795282/settings.txt)
[image]

theyosh, 2020-06-19:
This looks like a missing of corrupt logging.cfg. Or did you made a 'logging.custom.cfg' file? Change that name so it would not be used and try again.

Else, remove the logging.cfg and download the original once more: https://github.com/theyosh/TerrariumPI/raw/master/logging.cfg

theyosh, 2020-06-24:
Any updates? Found the issue? As I cannot think of something else

kaju666, 2020-06-25:
Hi

sorry for late reply i didint have logging.custom.cfg. 
logging.cfg was stock.
Finally i made full config from scratch so can't now check what more could be wrong.

closed 2020-06-25 by kaju666