theyosh/TerrariumPI #408: Proper use of notifications

darrynl, 2020-06-21:
Hi All.
Can someone tell me if my setup is correct?
What I am trying to achieve:  I want to keep the temperature at the sensor between 31 and 32 degrees celcius and be notified if the temperature goes above or below this.
I am controlling a heat mat via a relay connected to TerrariumPI.
Sensor Setup:
[image]
Environment Setup:
[image]
[image]
This appears to be working for the most part, but I am seeing temeprature fluctuations of over 2 degrees.
[image]
This is also causing notfications to be sent out each time the temperature goes above and below the limits I set.
Is my setup correct and that is just the way the system works, or have I misconfigured something?
TIA.

theyosh, 2020-06-24:
Hi @darrynl 

The notifications will 'spam' you :P Sorry, it is not that fine tune able. The notifications will trigger on the sensor, and not really on the environment logic. So yes, you could get about 2 or 3 messages that the temperature is to high, due to the time it takes to lower to the wanted values. So not sure if it will help you.

But your setup is right for 95% I think. What you could try is to put a value '0' in the Settle time timeout. That could make the software a bit faster to trigger.

Also, I would advise the increase the Limit min/max values. Because when the temperature drops outside those values, the system will see that as an error. I would use values like 20 and 40, just to be sure that some spiky values can be measured.

But this logic will just turn off and on the heating mat. The mat needs time to heat up, so when you power it up to start heating, it could be that for one minute the temperature is still dropping due to heating up time of he heating mat.
If that is a problem, you could try to use a min value of 30.5 It should work with decimals also. But then, you get a very narrow window which is valid. I do not know how my system will react to that.

And final, the high alarm part of the heating environment is not needed, as you do not have a cooling system/device. So only the low alarm values are needed in your case.

darrynl, 2020-06-25:
Hi @theyosh 

Thank you for the detailed reply.  I do appreciated it.
I will play around with this a bit more.

closed 2020-06-25 by darrynl