theyosh/TerrariumPI #410: Hardware support feature request for TE215 Moisture Probe

havealoha, 2020-06-24:
Please add support for the TE215 Moisture Probe

theyosh, 2020-06-24:
I think, you can use it, by selecting the 'ytxx-digital' sensor. And put the physical pin number as address.

It can only handle this kind of sensors, if you have the analog to digital converter between it. And therefore, my software will only register dry/wet. Not the amount of moisture. That is not possible, as the Raspberry PI cannot handle analog sensors directly.

havealoha, 2020-06-24:
I came to the same conclusion after writing some code to use as a "Custom script". There is a sensitivity dial on the TE215 but it is really finicky and not accurate enough for measuring moisture in a microgreens tray. It might be useful for a reservoir or leak detector in the digital out\PI. Maybe if like you suggest, I use an ADC I could get better readings but rather than use up all those precious GPIO's on an MCP3008 I could find a digital sensor. Can anyone recommend one for my situation? We can close this.

theyosh, 2020-06-28:
Hmm not sure how much soil you have, but I use: https://www.aliexpress.com/item/33017406335.html
Or: https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/ This sensor is less stable, but should work for at least 1 year.
But looking at your picture, they could be a bit to big.

Or when you are feeling brave: http://www.nodemcu.com/index_en.html
This board could probably readout the sensor using analog ports. And it has wifi, so it can be connected to your home network. And that would make my software be able to readout the sensor with the analog values without using all the GPIO pins on your Pi. But this will involve making Adruino code.

theyosh, 2020-06-28:
Also, have a look at: #412

theyosh, 2020-07-03:
Any updates? Can we close this issue?

closed 2020-07-04 by havealoha