theyosh/TerrariumPI #411: web service not accessible after changes config

blatogera, 2020-06-24:
Hello,

First of all, thank you for the great piece of software!
I'm having trouble accessing the web interface after a changes is made on the Settings - page.

I reinstalled like 15 times already, everytime a full reinstall fixing the issue, i can add sensors, webcam, all good.
but once I try to change the weater link, password, water and electricity price the webpage is not available anymore.
i can log in SSH, ping the Pi, but no webservice
i try sudo ./start.sh - web service starting but not accessible. no error anywhere... Can you please help?

Dave0408, 2020-06-24:
Hello @blatogera ,

in your ssh session check if there is a listener port open:
The command `sudo netstat -tulpn | grep 8090` should look like this:
```
pi@raspberrypi:~ $ sudo netstat -tulpn | grep 8090
tcp6       0      0 :::8090                 :::*                    LISTEN      492/python
```

Have you checked  `~/TerrariumPI/log ` directory? 
Can you verify that your changes are applied in `/home/pi/TerrariumPI/settings.cfg`?

theyosh, 2020-06-24:
@blatogera Hmm that is not how it should work..

So first of, in order to 'reset' the software, just remove/rename the settings.cfg file and restart. That will start the software cleanly. No need for a reinstall. Saves some time :)

Then next question: Did you install it with python2 or python3? Did you add the '3' to the installer commando?

Which kind or Pi are you using? A PiZero will give problems, and then it needs to run as the root user.

I need some more info to give you the right debug commands.

blatogera, 2020-06-25:
> Hello @blatogera ,
>
> in your ssh session check if there is a listener port open:
> The command `sudo netstat -tulpn | grep 8090` should look like this:
>
> ```
> pi@raspberrypi:~ $ sudo netstat -tulpn | [...]

image as attached, looking good. also in the log lie i found several errors not able to update the weather information, 
[image]

Found that the city I live, doesnt like the way its copied from the browser: 
https://www.yr.no/place/Hungary/Pest/Gy%C3%B6mr%C5%91/ - as it dont recognise the special characters in "Gyömrő"
sometimes gave good results on weather tab sometimes did not. and revert back to default.
since I set up some other nearby city it doesnt have the web service issue. but webcam is gone and need to add again after each reboot now.

blatogera, 2020-06-25:
> @blatogera Hmm that is not how it should work..
>
> So first of, in order to 'reset' the software, just remove/rename the settings.cfg file and restart. That will start the software cleanly. No need [...]

Hello, 
i tried for days with a Pi 1 B+ before i found here its not really supported. it was sometimes working but mostly not due to performace :D 

Right now I have a Pi4  -2GB and I installed with Python 2 i guess, didnt add anythig to the commands, only the regular install steps from the install section.

seems the weather location characters were using the problem, now it works with a nearby city set, 
but webcam is gone after each power cycle,  i could get a live image, and view it in "show" menu, but after reboot it1s not there.
also the settings.cfg does not contain the camera after reboot, only sensors and weather. maybe i do something wrong? Thank you

theyosh, 2020-06-28:
I am not sure. But I think it is better to do one more clean install. Use a Pi4, and use Python3. So run the installer with the parameter '3' to it. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35. This could also fix your weather issue. But what weather url are you trying to use?

Then next is the question, what did you enter for the webcam at settings? Without that information I cannot help.

blatogera, 2020-06-29:
> I am not sure. But I think it is better to do one more clean install. Use a Pi4, and use Python3. So run the installer with the parameter '3' to it. [...]

Hello,

the link I tried to put: https://www.yr.no/place/Hungary/Pest/Gy%C3%B6mr%C5%91/ which is with orginal characters: https://www.yr.no/place/Hungary/Pest/Gyömrő ( I used the first option)

For now it seems it started to work by itself without me doing nothing, however I'm affraid to put any of the upper weather links, now using a nearby city without special characters and it works fine, including webcam and the DHT22 sensor. Also without installing Python 3 . 
so Pi 4 with Python 2 with one sensor and one webcam has no issues ( just the weather stuff)

theyosh, 2020-06-29:
Hehe, utf-8 can be a bit hard to work with. But with Python3 I am able to use the following url:
https://www.yr.no/place/Hungary/Pest/Gyömrő

When it escapes to https://www.yr.no/place/Hungary/Pest/Gy%c3%b6mr%c5%91/ it fails to write a valid configuration file.

With python2 it could be that it will not work at all with those characters. But if you have place pretty close to it, it is fine. Only the sun rise and sun set values are important. The actual weather is just for fun. It is not used.

blatogera, 2020-06-29:
Thank you! sun rise and sun set times are the same, I will leave it as it is, cause it's working fine like that. :) thanks again!

theyosh, 2020-06-29:
Cool. Then I close this. Have fun with it!

closed 2020-06-29 by theyosh