theyosh/TerrariumPI #414: Can't Access Webpage

Zlo98, 2020-06-29:
First off let me say I have no experience with coding or raspberrypi so I may be asking a stupid question. I have not been able to access the webpage for my raspberrypi. I have installed TerrariumPi software onto my raspberry pi and I think I did everything correctly. When I try and connect to the webpage it trys to connect for a bit then says that the page took to long to respond. When I typed in "screen -r" it says that it has started a server at the http://{raspberry_ip}:8090 location. Any help would be greatly appreciated.

theyosh, 2020-06-29:
Hi,

if you see that message about that the server is started, you should be able to enter that address in your webbrowser. That should give you a web interface where you can setup it up.

Zlo98, 2020-06-30:
I keep getting a error saying the website is taking to long to respond no matter where I try and access it. I assume this is a problem on my end but do you have any ideas on how to remedy this?

theyosh, 2020-06-30:
Do you have VPN software installed? That could cause this problem. If so, try to shut it down and try again. If that solves it, look for a setting to exclude certain ip numbers or hostnames.

Zlo98, 2020-06-30:
The browser was refusing the connection so I had to add it as a trusted site. It connects instantly now. Super excited to get this set up and into the terrarium I'm building. Thank you for the software!

closed 2020-06-30 by Zlo98