theyosh/TerrariumPI #415: Temperature Alarm Max and Min changes after reboot.

ewiandr73, 2020-06-30:
Hi,

I have system running at the moment this is how they are setup.

### Sensor:
Temperature sensor: 1 Wire 
Alarm Min: 25
Alarm Max: 33
Max Diff: -10

### Environment
Light Mode: Timer
Day Start: 9am
Day Stop: 9pm
Time on: 720
Time off: 0
Power Switch: Light Relay

Temperature Mode: Sensor
Sensors: 1 Wire
Day/Night Diff: -10
Day/Night Source: Lights
Light State: Ignore
Door State: Ignore
Settle Timeout: 120
Power switch: Heater

The Problem I have is If I reboot the Pi after 9pm the temperature sensor Alarm values change and the snake get cold :(

### Sensor:
Temperature sensor: 1 Wire 
Alarm Min: 15
Alarm Max: 23
Max Diff: -10

It might have something to do with the time zone, I'm in Sydney Australia (AEST).

Regards

Andy

luke5sky, 2020-07-02:
I had the same behavior with just clicking on the save button in the configuration.

I think the original temperature is not stored when the system substracts 10 and does it every single time.

theyosh, 2020-07-04:
Hi sorry for my late reply. Am not sure, and thought that a reboot should not affect this. But if you have a temperature change configured in the environment, you should not change your sensor settings during the 'night' period.

Als @luke5sky is correct. I do not store the 'night' values in the settings. So the settings file will have the 'day' values. When it becomes night, I change the min and max values according to the environment setting. But this is NOT saved. This is done in memory. Long technical story.

So, that means when you reboot, the system will always start in 'day' mode, even during the night. But after 30 sec, the first run is done, and it should see that it should be in 'night' mode, and change the sensor values again. Because the day settings of the sensors are read, it should just lower once again to the right values. And it should lower and lower it more.

But then, it is hard for me to test this, as I have done it a couple of times, and never got the reboot issue. But then again, you should not reboot the Pi. I reboot it maybe once a month...

So, if you have a temperature difference configured in the environment setting, you should then never change the sensor settings during night time.. That is a 'known issue' (which I should add to the wiki still)

ewiandr73, 2020-07-05:
Thanks for the explanation.

The reboot normally is a result of a power outage as there is a lot of work being done locally to the mains power.
Why should we not reboot the Pi?

Andy

closed 2020-07-05 by ewiandr73