theyosh/TerrariumPI #416: Migrate from deprecated Python-DHT to CircuitPython DHT?

victorhooi, 2020-07-01:
I noticed that we're using the Adafruit_Python_DHT library (https://github.com/adafruit/Adafruit_Python_DHT):

https://github.com/theyosh/TerrariumPI/commit/f4fab35d447801c25ee535ac6a19c496e55d5ddf

However, this library appears to be deprecated since last year. The maintainers suggest moving to the CircuitPythons libraries instead, which is maintained - I found the DHT one here:

https://github.com/adafruit/Adafruit_CircuitPython_DHT

What do you think about migrating?

theyosh, 2020-07-01:
At the moment I do not have a development setup to do big changes. So this can be done when I have setup my development environment. Also, I would like to test with real hardware. So without that, big code changes will not be done.

Also, I have used more of the CircuitPython code, but they do not work flexible enough. For example I just tried an VELM6075 (UV light) sensor with I2C. And CircuitPython does not see the extra I2C busses I have created... So the software is a bit stuck to defaults. So only 1 sensor can then be addressed. That is not flexible for me.

So it depends on how flexible their code is. As their goal is to make it simple, not flexible.

theyosh, 2020-07-23:
As I am rewrite the code, I changed the library. So the new version will have the new library. The old version, I stick to what I have.

So I think we can close this?

theyosh, 2020-07-29:
Closing now. In the new rewrite version we will move to the new library

closed 2020-07-29 by theyosh