theyosh/TerrariumPI #417: Drop support for yr.no as weather source.

theyosh, 2020-07-03:
As the Norwegian weather service is changing their API, it is becoming not useful anymore. The API is much more complicated and will not give the information I need for my software.

https://developer.yr.no/doc/guides/getting-started-from-forecast-xml/

They removed the sunrise and sunset values. Which is HUGH important for my software. So that will involve more API calls to different endpoints https://api.met.no/weatherapi/sunrise/2.0/documentation#!/data/get

And you need to fill in lat and long values, which is not user friendly. And for sunrise and sunset values even a timezone is required. So that means a lot of headache with locations and time zones which an API should solve. And not by the developer.

So, after 4 years of using as a very nice and stable weather source I have to say good by to YR.no.

theyosh, 2020-11-14:
If there is somebody that does know a different source, post it here. As I would not rely on only one service.

I will also try to look into new weather sources

cyber-shifter, 2020-11-28:
I was looking into this on the OpenWeather Map Site. Digging through their FAQ (I was looking for a list of cities, with corresponding ID's) I found that their URL differed from the one you posted by a bit. They have:

api.openweathermap.org/data/2.5/weather?id=CITY_ID&appid=YOUR_API_KEY

Is this a recent change? I've tried inputting both your URL and mine, and neither seem to work.

theyosh, 2020-11-28:
No change. But I use the City name option: https://openweathermap.org/current#name

cyber-shifter, 2020-11-28:
Cool. I think maybe it isn't calling my city by the city name, but by the parish name. I'll take a look again when I get home.

cyber-shifter, 2020-12-08:
I figured it out, it is indeed calling my city by parish instead of city, but it works so that's all that matters. Thanks!

closed 2021-10-09 by theyosh

Jakewinkel91, 2021-10-15:
I need help with this lol, I had it working last night and then messed up my pi and had to start over, cant for the life of me figure out what i put in that worked!    https://api.openweathermap.org/data/2.5/weather?q=Manaus&APPID=  is what i have now thats not working, it says its not valid!

josh5555s, 2021-11-03:
> For TP3: `https://api.openweathermap.org/data/2.5/weather?q=[CITY],[COUNTRY_2LETTER]&units=metric&appid=[YOUR_API_KEY]`
> 
> For TP4: `https://api.openweathermap.org/data/2.5/weather?q=[CITY],[COUNTRY_2LETTER]&appid=[YOUR_API_KEY]`

TP4 url works for me, but not if I want to use &units=imperial. Changing the temperature type in the settings menu adds &units=imperial to the url, which results in the "not valid url" error, even though it was working before making the change in settings.