theyosh/TerrariumPI #420: Sensor data cleanup

theyosh, 2020-07-04:
After some years the data is getting big. So lets clean some stuff up.

1. Sensors data. Data older then 1 year is never used, so clean sensor data older then 13 months
2. Door data. Same as sensor

This should clean a lot. Because the sensor data will take most data. We keep all the switch data, so we can still calculate the totals.

Dave0408, 2020-07-04:
How do you store the sensor data? In what kind of database? A time series database like influxdb has something built in like 'Retention policy'.

theyosh, 2020-07-04:
As any good IoT software, I use sqlite. That does not need to install any kind of server. As alternative, I thought about rrd. That can also specify a max amount of data to keep.

But using a MySQL or any other server is overkill. So influxdb is not an option. Also, the installation is not that user friendly.

But the cleanup is almost done... need some small tweaks

theyosh, 2020-07-05:
In the contrib folder there is a script called `db_cleanup.py`. Run this script from the contrib folder with the same python version as TerrariumPI is running.

Make sure TerrariumPI self is not running when cleaning the database. That will cause data corruption.

closed 2020-07-14 by theyosh