theyosh/TerrariumPI #421: SONOFF multi switch

ewiandr73, 2020-07-05:
I have a SONOFF dual R2 switch that has 2 relays, I have loaded it with Tasmota firmware and both switches work via the Tasmota web GUI.  

I can get the 1st relay to switch on and off in Terrarium but not the 2nd one.

From the Tasmota commands the following is use but only switch the 1st really on and off.
http://<ip>/cm?cmnd=Power%20TOGGLE

This was the addressing used to address the individual relays.

Relay 1
http://<ip>/cm?cmnd=Power1%20TOGGLE

Relay 2
http://<ip>/cm?cmnd=Power2%20TOGGLE

I did try playing with the switches python file, but couldn't get it to switch the second.

I wondered if could use 'http://<ip>/cm?cmnd=PowerX' as the address but looks like anything after the <ip> is added by the python script.

Is there anything that I need to do to be able to switch both relays or does this feature not work yet?

Regards

Andy

theyosh, 2020-07-05:
Hmm I have not tested this. And also the code is missing a bit to support this. But that can be fixed. In the meantime could you try the following url:

 http://ip/cm?cmnd=Status%200

And try to paste the outcome here

theyosh, 2020-07-05:
Made an update so it is supported now. You can enter as address http://IP_OF_SONOF/#[SWITCH_NR]

So http://1.2.3.4/#2 will toggle relay nr 2 on the switch.

ewiandr73, 2020-07-05:
> Hmm I have not tested this. And also the code is missing a bit to support this. But that can be fixed. In the meantime could you try the following url:
> 
> http://ip/cm?cmnd=Status%200
> 
> And try to paste the outcome here

{"Status":{"Module":39,"DeviceName":"Tasmota","FriendlyName":["SW1","SW2"],"Topic":"tasmota_A538E0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T11:15:53","StartupUTC":"2020-07-05T10:55:16","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-07-03T08:38:05","SaveCount":90,"SaveAddress":"FA000"},"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"367/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]},"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"},"StatusNET":{"Hostname":"tasmota_A538E0-6368","IPAddress":"192.168.86.90","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"2C:F4:32:A5:38:E0","Webserver":2,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"192.168.86.81","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A538E0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30},"StatusTIM":{"UTC":"2020-07-05T22:11:09","Local":"2020-07-05T23:11:09","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:54","Sunset":"20:55"},"StatusSNS":{"Time":"2020-07-05T23:11:09"},"StatusSTS":{"Time":"2020-07-05T23:11:09","Uptime":"0T11:15:53","UptimeSec":40553,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"WiFi","BSSId":"E4:F0:42:F0:37:96","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:05"}}}

ewiandr73, 2020-07-05:
> Made an update so it is supported now. You can enter as address http://IP_OF_SONOF/#[SWITCH_NR]
> 
> So http://1.2.3.4/#2 will toggle relay nr 2 on the switch.

Thanks, I'll update, give it a try and let you know the outcome

ewiandr73, 2020-07-06:
Fantastic it worked :)

Thanks

closed 2020-07-06 by ewiandr73

theyosh, 2020-07-06:
Nice! Your output is what I expected, but it is not really helping me. So thanks for posting. And have fun with my software!