theyosh/TerrariumPI #422: Temp sensor adjustment

cyber-shifter, 2020-07-06:
I searched the closed issues for some information about this, but didn't find the answer I was looking for. Is there a way to adjust the readings that the sensors are feeding back to the system? I'm using BME280's and they are reading a little lower than actual temperature. The different is only about a degree or so, but I'd like to make the adjustment if it's possible. I was going to dig through the code myself, but I figured I would ask here first, in case it was a known thing. Please let me know if this is possible. I've also read that these sensors have multiple sampling modes, so I'm curious which one Terrarium Pi is using. Thanks!

theyosh, 2020-07-06:
Hi,

yes! I agree. I think I will add some kind of a calibration option. The most simple form is just adjusting the outcome with +/- amount.

Not sure how fast I will add this. Have to look ho much work it will be

theyosh, 2020-07-06:
About the sampling modes, I take the one that takes the most time. Hoping that this will give the best results... But I have not tested it much. And I did not really read into it. So I do not know what the different sampling modes do.

theyosh, 2020-07-06:
I found out that I do nothing with the sensor sampling. So I do not know.

But I have changed to a new BME280 library, and maybe it give a better readout? I did not pay that much attention on my temperatures. A few degrees more or less is for me not a problem.

So, you could do a `git pull` and re-run the installer. Then restart the software. At least the readout of the sensors has speed up a lot.

cyber-shifter, 2020-07-06:
I found an article on another git that has some details. I will try to get it at lunch. The link is on another machine. For me I would like it to be close at least, so that's why I'm asking. A few degrees of deviation is a bit much for me, but I'm going to try the new code first, once I get off work.

cyber-shifter, 2020-07-07:
Here is an example of what I was talking about:

https://github.com/letscontrolit/ESPEasy/issues/164

The weather/climate mode would likely be what would be ideal for our use case. But I leave that up to you. I'm going to try the new code anyway as well.

theyosh, 2020-07-07:
Hmm, as far I can judge, we are using that method: https://github.com/rm-hull/bme280/blob/master/bme280/__init__.py#L208

And I do not see how to change that in a external library as there is not enough options to change the method. The sampling parameter is the same for all the three values (temp, hum, pres). Where your link shows they use different. So that is something we cannot do. But then, the climate method is the one we use as far as I can judge...