theyosh/TerrariumPI #423: Unable to load web interface.

Fitzpa, 2020-07-06:
This is my first RPI project. I am using a Raspberry Pi 4 Model B Rev 1.1

I installed the Raspbian buster lite because I don't think stretch is available anymore, even though I know in the repo README it says to use Stretch.

I have ssh enabled and I am able to access my RPI from my desktop Windows machine remotely through SSH using the Git Bash terminal for Windows.

I followed the instructions in the README, installing git, cloning the repo, changing directories into TerrariumPI, running the installer script and then rebooting the RPI

On a separate monitor, I have my RPI hooked up and displaying that a server started on http://myip:8090, though when I try to access that from my desktop it doesn't load. In the network tab, it says `(failed) net::ERR_CONNECTION_REFUSED`, also the Type says it's a document which doesn't seem right but this is my first time messing with an RPI. 

I have tried using other browsers with no luck.

I have tried adding the URL to the list of trusted sites in my internet settings on windows.

After the RPI boots and says `Starting TerrariumPI server running as user 'pi' at location: http://myip:8090 ...` and appears to start and run fine, if I log in it says `run-parts: failed to stat component /etc/update-motd.d/05-terrariumpi: No file or directory`

I don't know if that is significant information. Really not sure what to try next. Please let me know if there is any more details I can provide to help diagnose this issue. Thanks!

theyosh, 2020-07-06:
Hmm,

Did you install it using python 2 or python 3? And could you look at: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi and try to make a screenshot?

Fitzpa, 2020-07-06:
I re-installed using `sudo ./install.sh 3` And then `sudo ./start.sh 3` And it worked. 

Thanks!

closed 2020-07-06 by Fitzpa