theyosh/TerrariumPI #424: How to get BME280 sensor working

Fitzpa, 2020-07-06:
**Setup:**
 - Python version: 3
 - Raspberry PI: 4
 - [BME280 Sensor](https://www.amazon.com/gp/product/B01N47LZ4P/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

**Problem**
I don't know how to find the address of my BME280 sensor or how to check if the sensor is connected correctly.

**To Reproduce / Installation steps**
[The resource I used to identify pins](https://pinout.xyz/)
VCC is connected to physical pin 1
GND is connected to physical  pin 6
SCL is connected to physical  pin 5
SDA is connected to physical  pin 3
CSB is connected to physical  pin 12
SDO is connected to physical  pin 19

Thanks!

Fitzpa, 2020-07-08:
I had typed in 0x76 instead of 76 for the address. That’s why it wasn’t working.

theyosh, 2020-07-27:
Hi, in the new version I will fix this... as it is confusing more people

blatogera, 2020-08-04:
Is it possible to use a 2nd i2c bus?
asking because for some reson i cant change the address of the BME/BMP280 (violet type) address, even resoldering the jumper pins, still its recognised on address 76 instead of 77, 
If i'd add it to a 2nd bus that would solve the problem right? @theyosh

theyosh, 2020-08-04:
Hmm, I am currently testing with 4 I2C busses :P So it is possible.

Look at: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README and search for 'i2c3' Here is described how you can enable another I2C bus at default pins: BCM 4 and BCM5 (https://pinout.xyz/pinout/pin7_gpio4# and https://pinout.xyz/pinout/pin29_gpio5#)

So make sure those GPIO pins are still free! As changing the pins for I2C is not always working for me. So I keep default pins.

After changing the /boot/config.txt like 

`dtoverlay=i2c3`

and a reboot, you should be able to see another I2C bus with i2cdetect -y 3

And then you can just hook up the BME280 to the new bus.

When entering the address, use '76,3' in the sensor settings form, where ',3' is specifying which I2C bus you want to use.

I do it also this way. As I noticed that changing the BME280 address is not always working

blatogera, 2020-08-04:
@theyosh Thank you! Working great now!

theyosh, 2020-08-08:
Ok nice! Closing second time ;)

closed 2020-08-08 by theyosh