theyosh/TerrariumPI #425: [FEATURE]: Support of more than one EG-PM2 power switch board

Neamh, 2020-07-09:
As stated above, it would be nice to support more than one of the EnerGenie EG-PM2 switches.
With the sispmctl tool it is possible to control all attached power switches. Discovery of attached switches is also possible.

--> https://sourceforge.net/projects/sispmctl/files/sispmctl/sispmctl-4.7/

theyosh, 2020-07-09:
Hi, yes, I agree! Problem right now is that I do not have a working development setup. Working to fix that.

So I have to see if I can make it work in a simple way. And then later use the discovery option. Because that would be nice to have also. And I do not have two of those devices, so I do need some feedback from you when I make some code changes.

theyosh, 2020-07-09:
I also see that I do not read out the state of the switches, which can also be done. So that means that my software can also update it's state when you toggle the power switch outside my software... nice!

theyosh, 2020-07-13:
Hi,

could you do a small test for me? If you have connected both switch board, could you run as normal pi user:

`sispmctl -s`

and past the output here? I am interested in the serial numbers it produce

Neamh, 2020-07-13:
Sadly, my second board is currently used on another terrarium not connected to my Pi.

The output looks as follows:

pi@raspberrypi:~ $ sispmctl -s
Gembird #0
USB information:  bus 001, device 004
device type:      4-socket SiS-PM
serial number:    01:01:5c:3d:d7

theyosh, 2020-07-13:
Hmm, ok, I was hoping on output of two boards. So I am not sure if my code will work well with two boards. I can't test it.

But I think it will work. Also scanning for connected boards is done.

The downside of this update is, that your power switches will be seen as new switches, and will be added twice. This is due to the fact that the addressing of the switches is slightly changed, and there will not match with the old ids.

So, when you update the code and restart, you will have to delete the old switches, and configure the new ones. Else you will always have double power switches after a restart.

`git pull` and restart.

theyosh, 2020-07-23:
Hi, any updates on this?

Neamh, 2020-07-24:
Sorry, currently busy with building my new rainforest terra...
I did a clean install on a fresh Raspian Lite  but somehow there were some error messages scrolling fast through at the end of the installation and then disappears. Afterwards it asks to reboot as normal.
Finally, there was no MOTD after login, it didn't start as a service and the start script didn't work neither.

Neamh, 2020-07-24:
OK, did a reinstall and at least it is running. Still no MOTD and there was this error during install:
[image]

Furthermore, the EG-PMS2 is not recognized. This is the error in the log:

2020-07-24 18:06:27,278 - ERROR   - terrariumEngine      - Engine loop: Power switch has problems: Command '['/usr/bin/sispmctl', '-d', '-1', '-n', '-g', '4']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 639, in __engine_loop
    self.power_switches[power_switch_id].update()
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 231, in update
    data = self.get_hardware_state()
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 476, in get_hardware_state
    data = subprocess.check_output(cmd).strip().decode('utf-8').split('\n')
  File "/usr/local/lib/python3.7/dist-packages/gevent/subprocess.py", line 404, in check_output
    raise CalledProcessError(retcode, process.args, output=output)
subprocess.CalledProcessError: Command '['/usr/bin/sispmctl', '-d', '-1', '-n', '-g', '4']' returned non-zero exit status 1.

Neamh, 2020-07-24:
It seems it uses the wrong device id. The first device is called Gembird #0 

`sispmctl -d 0 -n -g 4 `
gives the status of outlet 4 on the command line.

Are the EG-PMS2 recognized automatically or do I have to add them manually? If manually, what is the syntax in the UI for 'address'?

theyosh, 2020-07-27:
Ok, 1 thing at a time. The installer issue is not an issue. So please ignore. Also your MOTD issues is now ignored as that is not related to this.

The new code should be able to find all the connected USB power switches. So if you have 2 board connected, you should see at least 8 power switches after startup.

Then, the address numbering is: [Powerswitch nr (1-4)],[Board nr(1-x)] where the Board nr is optional. When not entered the value '1' is assumed. And I do know the software starts counting at zero, but humans do not. So I expect the board and power switch numbering always starting at 1.

But do you have now two devices connected? Did you had more powerswitches due to auto scanning?

Neamh, 2020-07-27:
So, actually, there is only one EG-PMS2 board connected and it is not found automatically. Furthermore, even when I try to add the switches manually using the adress format you provided, it didn't work.

BUT: I found the solution for all the mess. Your script is calling /usr/bin/sispmctl. Called from commandline this didn't work on my Pi. I guess this is the one from Raspbian repo which is far too old. It didn't even recognizes the EG-PMS2 at all.

I took the source code from Sourceforge and compiled a recent v4.17 of sispmctl to /usr/local/bin/sispmctl.
After exchanging the files and restarting, it now finds the connected switch board and added all switches correctly.

I hope, I will get another EG-PMS2 soon to test the multi-board function properly.

BTW: MOTD is now working after restart ;)

theyosh, 2020-07-27:
Ah nice. This helps a lot. So I will change the code, that I will look for more places where sispmctl is available. And then the version in local will then prefer.

theyosh, 2020-07-28:
Made an update... could you test once to see if this works for you?

Neamh, 2020-07-28:
Yes, it works, Is it intended that the address is now [switch#],[serial number] ?

theyosh, 2020-07-28:
Yes and no :P

I guess that getting the serial number of the devices is a bit hard. Need some commands for that. So I accept also numbers from 1 - X for devices, and during loading, I (software) change that to the corresponding serial based on a listing.

So, yeah, after auto discovery / or when you manually enter the address, it will be updated in the form you mentioned. So when you reorder the USB devices, the switches should still work correctly.

You can also enter the serial your self, but I thought to keep it simple for the customer :) https://en.wikipedia.org/wiki/KISS_principle

theyosh, 2020-07-28:
Also, when you only have one device, you can even leave out the device number/serial....

Neamh, 2020-08-08:
Finally, I got my second EG-PMS2 and it's working as expected. Both boards are recognized and every outlet can be switched. Good job, thanks.

closed 2020-08-08 by Neamh