theyosh/TerrariumPI #426: sht20

thegrumpyoldman, 2020-07-09:
I want to add a sht20 to TerrariumPI
I have installed the library raspberry pi sensor
with i2cdetect -y 1 I can find address 0x40
also I can read the sensor
the problem is that TerrariumPi does not give a reading
any sugestion?

theyosh, 2020-07-12:
What address did you enter in my software. It should just be 40. If that does not help, could you share your script? I am not able to test this sensor at this moment....

thegrumpyoldman, 2020-07-12:
I use 40 and on the moment it is running
is there documentation how to control temp 
and rh?

theyosh, 2020-07-19:
Hi, sorry for the delay.

But you could do this in the environment part, and then in the temperature section. I guess you have a heater, so use the low alarm tab to select your power switch. In the general heater section you can select the temperature sensor.

The min and max alarm values are use for triggering the power switches

theyosh, 2020-07-23:
Is my last message clear? Have you been able to setup your environments?

thegrumpyoldman, 2020-07-24:
yes understood message
because we want different times of light i am writing my own program

thanks Jos

theyosh, 2020-07-24:
Ok, then I close this issue now.

closed 2020-07-24 by theyosh