theyosh/TerrariumPI #427: [BUG]: Error aftrer reinstalling

Kollimat, 2020-07-12:
Hello I reinstalled TerrariumPI after my power goes down in middle of update and now i have this error every time.
[image]

theyosh, 2020-07-12:
My mistake. Do a new `git pull` and it should work again.

But if you did a re-install, I suggest to switch to Python3. Which is more supported and has more sensors. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

Kollimat, 2020-07-12:
Perfect, Thank you, all is now function. Is there a way i can support you, like buy you a beer or coffe ?? :)

theyosh, 2020-07-13:
Yes you can! :P https://github.com/theyosh/TerrariumPI/wiki/Donations

But also, if you did not post pictures at #210 then that is also an option :)

theyosh, 2020-07-19:
Closing as it is working again...

closed 2020-07-19 by theyosh