theyosh/TerrariumPI #428: Email notification issue - ( twitter help )

blatogera, 2020-07-13:
Hello, I'm trying to set up the notifications either via email or twitter,
After input the email address,  SMTP settings, username, password, and selecting the desired notofications still nothing comes.
rebooted the system, settings were still stored, but does not work.
tried with gmail and other email provider too, with their settings but so far no luck.

I tried to give a go on twitter but as i'm just a NOOB, I've no idea how to obtain Consumer token, secret, Access token, secret.
Any help would be truly appreciated - checked the wiki but did not find relevant information on this.
Thank you!

theyosh, 2020-07-15:
Getting twitter tokens: https://iamstarkov.com/get-twitter-tokens/

For Email, I can't judge why it is not happening. Could you post a screenshot of the notification settings? Make sure you clear your password before posting the screenshot.

blatogera, 2020-07-15:
hello,
[image]

I'll try the twitter thing too, thank you

blatogera, 2020-07-15:
also the twitter thing page changed a little, i can get this, 
I assume the API key/secret is the consumer key/secret and the Acess token/secret is clear there
[image]

theyosh, 2020-07-19:
Hmm, I think you should have to go to the 'Access Token & Secret' part. But I do not have twitter myself. So I do not know how it looks like.

I just use the general way of connection applications. So it should still be possible to connect to twitter.

blatogera, 2020-07-22:
seems none of the push services works for me, no email, no twitter ( the rest I dont use)
is there a way for updating the Terrariumpi keeping all the configs?
maybe install python 3 along the way? still keep all settings?
I have a feelng something was not installed correctly

theyosh, 2020-07-23:
Hmm, I do not use those 'free' services. So I cannot help much more.

For Gmail, I think you need to set an option to allow SMTP/IMAP connection. Could it be that this is not enabled. I do not know where it is, Google for it.

For twitter, it should work, I think.

But have you selected a message to be sent through a service. When you have configured mail, twitter etc, you also need to select which messages you want to receive.

Debugging is described here: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

theyosh, 2020-08-27:
Any updates about this?

blatogera, 2020-08-27:
No, couldnt make it work either twitter nor gmail.
but as I'm he only one having trouble with this, i guess the problem is on my setup somehere :)
you can close, i'll try a few weeks later once will have some time.

theyosh, 2020-08-27:
Ok,I hope at least Gmail connection will work. Twitter will just 'spam' public tweets. So it does not support DM's

closed 2020-08-27 by theyosh